misan/dcservo #2: dcservo_pro_micro doesn't compile

smellsofbikes, 2016-02-14:
Arduino IDE reports:

arduino_pro_micro_servo_with_serial_interface.ino: In function ‘void countStep()’:
arduino_pro_micro_servo_with_serial_interface.ino:109:23: error: ‘PINF’ was not declared in this scope

I think replacing 'PINF' with 'analogRead(A0)' might fix this?  For speed I'm going to change dir to digital 10 and do a digitalRead of that and see if that works.

misan, 2016-02-14:
Have you configured your IDE to use the Arduino Pro Micro board (based on 32u4 not 328) that has different ports than Arduino UNO or Mini?

smellsofbikes, 2016-02-16:
Ooooh, sorry, you're right.  I thought what I had was a pro micro, but it has a 328.  (That's what I get for buying ebay specials!)

closed 2016-02-16 by smellsofbikes

q2dg, 2016-04-04:
Maybe in Readme.md should be specified what kind of Arduinos are compatible (328-based only?). I mean...ARM are not, right?

smellsofbikes, 2016-04-04:
More specifically, Atmega32U4, because it needs more hardware interrupts than the older 48/168/328's possess.  That's what bit me: I was confusing micro with mini, which has a 328 and lacks the interrupts to run this.
The code works great, by the way, although I'm finding it challenging to tune with a fast motor.

sanchosk, 2016-04-04:
Try to modify the code for tuning - go for X1000 instead of X100 - worked for me very well with faster RPM motors.

misan, 2016-04-04:
@q2dg  I am using it for ARM for example with MKR1000, have a look at this project: https://www.hackster.io/misan/control-the-position-of-a-continuous-rotation-servo-via-wifi-270e64?ref=user&ref_id=44495&offset=0

misan, 2016-04-04:
@sanchosk It is difficult to have a one-size fits all solution. For anything else: "use the source" :-)

misan, 2016-04-04:
@smellsofbikes Hi John, high speed motors (or high count encoders) can create many more interrupts per second than my code can handle leading to a bit of a mess. That is why I did a Maple Mini version (available on youmagine site) that works over an ARM Cortex M-3 that is a cheap platform and can reach where Arduino could not. Anyway, magnetic encoder removed the need needs of encoder interrupts altogether but again I am not sure what is the max speed these sensors can work at.

sanchosk, 2016-04-04:
@misan Sorry for the misunderstanding, I was replying to @smellsofbikes. However, I sent a pull request for the modified pde, too.