misan/dcservo #6: How to use PID tuning tool?

dvjcodec, 2016-04-26:
Hi,
I try to understand how to use PID tuning tool, but can not :( Start, I was open sketch in Process3. Then change usb port. In Arduino Uno I have "dcservo" sketch. When run sketch into Process3 my DC motor forward and backward, but not stable.
How can I change PID values?

sanchosk, 2016-04-26:
Simply - use the key P for increasing the proportional part, key p for decreasing it. Similarly use keys I/i for integral and D/d for derivative part. Try to get the movement as smooth as possible with as little vibrations as possible.
Your goal is to make sure the motor moves to the target position, slows down and stops exactly at the spot. The faster it gets there, the better.
My values varied a lot - for one motor I had P=4, I=4, D=0.25, for other one I had P=13, I=16, D=0.07. For the second motor - if I increased D parameter, the motor made quite ugly noises when it started to "slow down", so I had to be a bit more conservative ;)
Good luck!

Jxplicits, 2016-04-29:
Hi, guys
I'm trying to get a nice curve out of it but is this a bug or do I have something wrong with my setup? 

[image]

English isn't my main language sorry if there's any error.

misan, 2016-04-29:
Tuning tool was upadated a while ago to allow taller steps. I have not tested since then, it seems there might be have an issue but your output looks as if the vertical scale was suffering a modulo operation. I will check it out tomorrow to see if I can help.

misan, 2016-04-30:
Hi again, I have checked everything and I can see why you get this: the provision of location recording while moving on the Arduino is a single byte per sample, so when reporting back with the S command any distance larger than 255 is reported as D mod 256 :-( 

I have added a now a warning in the code. Of course if you want Arduino might use 16 bit value instead (at the cost of doubling the storage required). Or you can just see that the graph you see has suffered this modulo operation and what ends on the tops continues next on the bottom of the graph.

If you use a targetVal < 220 it should be mostly ok.

Jxplicits, 2016-04-30:
Hey, I'll try it later this week. I think I've found the correct values by trial and error but also I want to check what I can get with this tool. 
Thanks a lot for your help and your work  :)

dvjcodec, 2016-05-29:
Thank you. I will test tomorrow.