misan/dcservo #12: little modification

Akex2, 2016-09-24:
Hi Misan,
I'm trying to use your arduino sketch, but I want to compile it for Arduino nano or Arduino pro. 

And on my h-bridge board:http://www.ebay.fr/itm/Dual-Motor-Driver-Module-Board-H-bridge-DC-MOSFET-IRF3205-3-36V-10A-30A-TE589-/381782113555?hash=item58e3fcbd13:g:X9IAAOSwmLlX4uGE
 there is only one PWM input, and one "pin dir". So to make the motor turn clockwise or reverse either I activate the pin dir, or I deactivate the pin dir. Is it possible to do this ? and how can i do this please ?
Thanks !

misan, 2016-09-24:
Have a look at how something similar is done for brushless motor https://github.com/misan/dcservo/blob/master/dcservoESP_magencoder_brushless.ino#L125

In you case last command should be analogWrite(PWM, abs(out)); though. Of course PWM and DIR pints would need to be defined within your code as output pins.

coupmstr, 2016-11-20:
If your still struggling with this I have a demo code for this board that will help you uncerstand it

misan, 2016-11-20:
Hi coupmst,

If you are willing to share maybe a link can help other people

misan, 2016-11-20:
Sorry, I press the wrong button :-)

I meant that coupmst, please feel free to share a link here,

coupmstr, 2016-11-20:
https://drive.google.com/file/d/0B-OvshlKUhh3Z1Bubk5PMERETVU/view?usp=sharing 
Best way I can find to share this one, I had issues working out how to work it