misan/dcservo #14: Feature Request: PID autotuning

lonelymyp, 2016-11-09:
It would be great to add to the program algorithm for the automatic calibration of PID.
Similar to the way it is implemented in Marlin.

misan, 2016-11-09:
I do agree. But when I tried with the PID autotune library I have got no useful calibration (it uses Ziegler-Nichols algo https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method ) http://playground.arduino.cc/Code/PIDAutotuneLibrary

But I am open to use any other algorithm that might work ok in this environment.

pppalain, 2017-12-07:
The Ziegler-Nichols algo would provide a slow tracking. It would be good to find critical KP.

a tuning which works well for me for a 3d printer setup / step tracking finding the critical KP by setting the KI and KD to zero then increase kp until critical unstability is found. for motors it would be back and forth with a certain amplitude but not increasing or decaying.

Then set KP to be KP critical /2
then start KD very small until instability starts and then divide this by two
after kd is set, kp can be increased and then kd can be increased until unstability.  The system will become more stiff.

then use a very large ki for quick tracking.
remember that only ki can reduce the error to zero.

in the control zone (abs(output/kp)<255) ki is large else, ki=0 this prevents integrator windup
at the moment I am unable to test an algorythm since both my axis have melted (pla with hot motors)  I will reprint with ABS and resume.

so in my system I have a 
kp =10 
ki=200
kd=0.05

pppalain, 2017-12-07:
oh in my system my kp critical was 15 so it was increased after kd was added.