misan/dcservo #16: DCservo for milling?

joachimottmer, 2016-11-15:
Hi Miguel,

a couple of weeks ago, i tried out your work with a linearsystem of an old colour printer and it wokred well. It is fanstastic -  Now i am thinking about building a cnc Mill instead of an 3DPrinter with your servosolution.

And here is my question:
When marlin or chillipepr is sending steps and direktion to your arduino-servocontroller, I gues it is sending an amount of step signals in a certain direction in a specific time. 
But it gets no feedback, and though it does not know whitch step the motor is performing and with whitch volocety it is mooving.

In case something is going wrong, and the linearsysthem is sticking somewhere, the PID-regulator would strengthen the power of the motor, to reach the position the axis is heading for.

When the resistence is overcome, the axis will move much faster to the defined position, than my feedrate would allow. 

For a printer not a big deal, for a mill a problem.

Is there a posibility to let marlin or Chillipepper or maybe Keeflop knwo, that the DC-Servo is behind its job and than break down the whole work in process?

Is the DC servo following the commands from the CNC -Controler step by step? and is there a posibility to make the motor follow just one step with maximal force? - just like a steppermotor?

What is the servocontrol doing with the steps, which it could not process in the demandet time?
does it have  a buffer or memory so an intelligence is working here, of which the cnc-controler(Marlin oder Chillipepper) has no clue of?

How could i control the maximum speed of the processed command to take care of the problem to exeed the maximal feedrate of my millingjob?

I know, these are a lot of questioins and maybe I am asking the wrong person, because you did this whole project not for milling but for printing - 

Anyway, I am so impressed about your work, that I would love to use it instead of using Steppermotors.
Just because it is so cool, what you accomplished.

Great great respect:

And If you would have an idea for me - wonderful. 

Thanks a lot 

Greetings from Munic

Joachim

misan, 2016-11-15:
Hi Joachim,

With a position-only control loop you are right: if blocked it might speed up above the feedrate as previous steps have been so far missed. But you can avoid that if you add a velocity/speed control loop too. That way you may limit the max speed even when position control wants the system to move quicker. 

Many commercial servos will give up (raise an alarm output) when the following error is above a certain (user definable) threshold. That will effectively pause the system (a CNC for example) till the blockage cause is removed and the operation resumed.

Please note that in any combined motion (more than one axis) the problem may affect one motor but no the other, so you are in trouble is motion progresses any further. Speed is only one of your problems but maybe not even the more important.

If you control all the motors with the same processor, all the information about the error on each axis is available for you to handle appropriatedly. However, if each motor is controlled by a different processor you may need to find a way for any type of reporting back to the main system you need. 

Current code attempts to move the motor anytime there is any error, even if a single step, though usually due to motor deadband, a small error like that may go uncorrected initially (maybe later it will be corrected if enough integrative gain).

Best regards,

Miguel
P.S: Sorry if I am missing any of your questions.

joachimottmer, 2016-11-15:
Hi Miguel,

thanks for your answer, 

it helped a lot, but regarding one thing, I still do not understand.

how is your Servocontroller handeling the step-direktion commands comming from marlin?

when one step with direktion is beeing send from marlin, what happens?
Does the servocontroller changes the PWM-Signal of the commanded direction and doublecheck the input of the encoder, weather the commandet position is reached? and than processing the next step?

or does the Servocontroller counts the commanded steps in one direction as an absolute lenght and than outputs the PWM signal till this commanded destinationpoint is reached?

Would it  be possible, to "make" the controller  handle every step for itself with the full possible motorpower, and be linked to the stream of steps coming from Marlin?

When I look at it Step by Step, I would have just a very little change from one the the other.
And even the time between these changes is a small amount- how could i than tell the PID controller to give the maximum Power to fullfill these changes??

Thanks again

All the best Joachim

misan, 2016-11-15:
Hi Joachim,

Every time step input gets a pulse it updates the target value (increases or decreases by one depending on the value of direction pin).

Whether a new pulse or not, each millisecond the PID code is run and there is chance for PWM output to be updated depending on the following error.

The relationship between error and out is mostly the P gain in the PID controller. 

A high P value will mean max power will be applied to the motor even for a small error (like a single step). However, such configuration is most likely producing an unstable control loop (that won't be able to settle once reached the setpoint) and it will tend to oscillate around the setpoint.

joachimottmer, 2016-11-16:
Hi Miguel,

thanks again for showing me the direction in which I now have to go....
You are right, a static high P gain can be the solution. Even if I would turn on the Mill and sent the axis to its 0-value the whole system would break....

Maybe I could change the PID values, to a step by step optimisation, when the machine-axis had reached their 0- Positions????

Anyway - than would only make sense, if I could trigger an emergency-stopp in case a certain axis is a defined amount of steps behind the position it should be.

But than...? how would I know which axis had triggered the Emergencystop?
And how could I tell Marlin or chillipeppr what happend?

But OK, why should I expect the cool way to be the easyest?

What ever I will find out - I will let you know? 
If you are by chance in Munic, let me know. 
0049 - 177 - 3259437

Al the best and thanks again

Joachim

misan, 2016-11-16:
Hi Joachim,

You can definitely change P value on the fly, or make the code handle a different P value depending on the magnitud of the error. 

On the other hand, if each axis provides a different ALARM signal, you know which one is to blame if your motion controller processes each one of those inputs independently. 

Alternatively, what some CNC machines do is to lower the feedrate whenever the following error of an axis grows above a certain threshold. For that you'd need another signal requesting a feedrate reduction to the motion controller per axis.

Chilliepper may not be the tool to warn as it is mostly a g-code sender, which ample buffer space, so any reaction on its part may be too late for your needs. Marlin seems to be your motion controller so there is where you'd need to add the alarm and feedrate feedback inputs, plus the corresponding code. I reckon there is a pause function, that might well be used by alarm inputs while movement feedrate maybe lowered by the feedrate feedback inputs.

joachimottmer, 2016-11-19:
Hello Miguel,

I read, I  learned and I unerstood - Chillipeppr is not the solution, but the motioncontroller behind it.

There is GRBL, Marlin and TinyG2. And as i was looking for information, I came across this discussion.
https://groups.google.com/forum/#!msg/devtinyg/rhPChJ2k4F8/4l2esLddAwAJ

There Rob Giseburt and Alden Hart are explaining how  TinyG2 is processing a Gcode command.

First Rob Giseburt:
"Let me very briefly explain how our system works:
1) We ingest moves in the form of gcode, parse it, process it, and hand it to the Planner.
2) The Planner puts the move into the move queue, and then stitches the moves together so they will execute cleanly.
Separately, we have the interrupt/timer driven run loop which takes the moves off of the move queue and executes them.
We break the moves into up to three "sections": head (acceleration), body (cruise), and tail (deceleration). Not all moves will have all three sections.
Each section is broken into "segments" of time that are all the same velocity. Each segment is at minimum MIN_SEGMENT_MS milliseconds (currently 0.75) long.
The "exec" function generates the sections, which are then each prepped for the loader. The DDA/stepper timer interrupt then calls the loader when it's out of steps, and the loader prepares the structures for the DDA/stepper interrupt."

Läter Alden Hart:
"- Part of the stepper system is a "feedback" counter that counts output pulses. See encoder.h/.cpp..........The encoder feeds error terms back to the step generation routines in case there is an error between model space (work coordinate system) and the joint space (steps). If the math is working correctly (which it always does) the error term is less than 1 step. The system keeps track of these error terms and a bunch of other low-level data.  ..............See here  https://github.com/synthetos/g2/wiki/Diagnostics"

Now my question: Would it be possible, to sent, - instead of the counting of output pulses - 
 the actual encoder values to the "feedback-counter" of TinyG and copy the TinyG original- feedback stepcounter to your arduino-servocontroller, so that it counts the step and direction values, which it is processing before it sents them to the internal process of creating  pwm-signals for the motordriver?

So, my simple thougt without any knowlede - the tinyG board woud be the ony board in the cycle whitch is actualy reacting to any feed-errors that may turn up.

The servo-drive than only had a possible "itself-correction-time" of one millisecond, whitch would be in my system at a slow feedrate almost nothing, at a high feedrate of 5400mm/min only 1/10 of a millimeter and at even higher feedrates  it would be irrelevant because this would be the speed of a rapid move: "home to zero" without milling.

Even with a high proportional value of the PID-Controller it could be less likely, to have a alternating, swinging system???

I am asking this, because if have not the experinence jet to really check, weather my thought is total bullshit. So your quick assessment could save me tonns of time an frustration;-)))

Thank you  for helping me

all the best

Joachim

misan, 2016-11-19:
GRBL or TinyG2 do a similar job, which is the one of a motion controller (plus they both interpret g-code). While there is a significant difference in their level of complexity (s-curves vs trapezoidal speed profiles) they both create the right timing for step and direction signals on the various axis.

My advice is to use any of them connected one of more PID-controlled servos and see whether you are happy with the preliminar results or not. 

I am not familiar on the type of encoder-based feedback TinyG2 is capable of, so I cannot help you in that front now (though it might well be a suitable solution if you plan to use steppers closed-loop). 

Any of these boards can handle a kill-job signal that might be driven by any of the motor drivers having trouble following the positioning commands on any axis.

Just my two cents,

Miguel

joachimottmer, 2016-11-19:
I am thankful for every cent !

and the advice: make and see  allone is worth  a lot..... 

Thanks again 

Joachim

joachimottmer, 2016-12-05:
Hi Miguel,

i have a strange question: My cheap Arduino Nano from China made a good job with your code - in the begining. 
I bought a bigger Motordriver from China  and  a 100W motor - everything worked well - but

suddenly, the motor war running without stopping anymore.

I found out, the nano is counting up targetvalues, even, when there is nothing connected???

Sometimes the positionvalue is climbing as well - somtimes it stays fixed - Again nothing connected to the board instead of the USB-Connector to my computer.

When i take a new arduino Nano - no such things are happening - with no connection the targetvalue stays 0 and the Positionvalue as well.

Do you have an idea, what I did wrong? 
I guess the strange values are due to a damage at the nanoborad - but i don't know what the reason is???

Do you have an idea?

thanks a lot

Joachim

misan, 2016-12-05:
It sounds like internal pull-ups were somehow damaged, A pin not wired to
anything may pick up ambient noise and register state changes, often at
power-line frequency.

That's my guess.

joachimottmer, 2016-12-07:
Hello Miguel,

just to let you know: Again, you were right!

I am using these very very cheap "aliexpress nanos" and I am not sure, wheather they have such a internal pull-up - but if, I have no idea about the quality.

Fakt is, that I followed your advice and added a 10kOhm resistor to Pin dig3 whitch is receiving the step signals and linked it to the 3,3 Volt output of the board.

And guess - Everything is just working again - No changes of the position value anymore ether.

right now, I am just playing arround with a 100 W DC Motor with this Aliexpress Motordriver (https://www.aliexpress.com/item/12V-24V-36V-450W-H-Bridge2-Channel-PWM-DC-Motor-Driver-Module-Reversing-Brake-S/32677902452.html?spm=2114.13010608.0.0.oQaF5N)
and quadrature encodes from old printers.

I am amazed - and if i will manage to make my mill working with these stuff, I will sent you a video
Thanks again for happy hours watching my Motor work like it should

sincerely

Joachim

misan, 2016-12-07:
Great news then. 

That H-Bridge seems poweful. A cheaper (but I am sure if better) alternative use BTS7960 for up to 43A @45V.

I am eager to see your mill working soon.

Kind regards,

Miguel

pfloutch, 2018-05-02:
Hello Joachim, 

a long time ago you tried to integrate Misan´s Software in you milling project.
Have you been successful?

Thanks a lot

Greetings from Allgäu, Germany
Gabriel

joachimottmer, 2018-05-03:
Dear Muguel,

sorry, that I did not give you the feedback you deserve for your great work.
I made some progress, but was not really happy with the velocity
I ordered different an faster bords but meanwhile I found this  JMC Servo

https://de.aliexpress.com/item/180W-NEMA-23-0-57N-m-Integrated-Servo-Motor-36VDC-3000rpm-JMC-iHSV57-30-18-36/32714727077.html?spm=a2g0x.search0104.3.7.16f54c3ayQzOcz&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10152_10321_10065_5722917_10151_10344_10068_10342_10547_10325_10343_5722817_10546_10340_10341_10548_5722617_10698_10696_10194_10084_10083_10618_10304_10307_5722717_10059_5711217_308_100031_10103_5722517_10624_10623_10622_10186_10621_10620_5711317,searchweb201603_32,ppcSwitch_4&algo_expid=b27603bc-36f8-47c1-894d-a46e8f9b50c8-1&algo_pvid=b27603bc-36f8-47c1-894d-a46e8f9b50c8&transAbTest=ae803_1&priceBeautifyAB=0

https://www.youtube.com/watch?v=NH8zH3fv0PQ

It is really cheap, so I decided to order one. I will go on with the
projekt on August 2018 because I have just to much stuff arround.

Next to chilipeppr I found this software and Breakoutboards but I am not
clear jet, if I would use it.

http://dynomotion.com/software.html

Sorry again for being that lazy -  but as soon as I start over, I will sent
you my experiences

Best regards from Munich

If you are planing to come to Germany, give me a call -
If I could held you in any other issue, please let me know.

You did help me and others so much -  I appreciate your work and you
enthusiasm

and I am really happy to hear from you, I feel sorry, that I "could" or
better have not been of greater help for you  !!

I will do my best hopefully soon,

sincerely Joachim

misan, 2018-05-03:
> Integrated-Servo-Motor-36VDC-3000rpm-JMC-iHSV57-30-18-36/
> 32714727077.html?spm=a2g0x.search0104.3.7.16f54c3ayQzOcz&
> ws_ab_test=searchweb0_0,searchweb201602_2_10320_10152_
> 10321_10065_5722917_10151_10344_10068_10342_10547_10325_
> 10343_5722817_10546_10340_10341_10548_5722617_10698_
> 10696_10194_10084_10083_10618_10304_10307_5722717_10059_
> 5711217_308_100031_10103_5722517_10624_10623_10622_
> 10186_10621_10620_5711317,searchweb201603_32,ppcSwitch_
> 4&algo_expid=b27603bc-36f8-47c1-894d-a46e8f9b50c8-1&algo_
> pvid=b27603bc-36f8-47c1-894d-a46e8f9b50c8&transAbTest=
> ae803_1&priceBeautifyAB=0

Hi Joachim,

No worries ;-)

I know these servos and I think they have such a good price it may not be
worth tinkering with large DC motors and powerful h-bridges and control
logic, as no money could be saved. Even better, the motors are brushless!

But there is some beauty in the DIY and the room for endless customization.

I will let you know when I go to Munich, thank you.

You can use GRBL firmware on an Arduino UNO for a basic controller. Then
one you mention seems more powerful (and expensive I guess).

Good luck with your project and best regards,

Miguel

sanchosk, 2018-05-03:
Just FYI:
https://www.youtube.com/watch?v=QCN7wJV224A

misan, 2018-05-03:
Thanks, Robert, it is a cool project but it shows that even doing it all
yourself you cannot beat the price of the servos Joachim was pointing out
;-)