misan/dcservo #17: step pulse duration

Cdriko, 2016-12-12:
hello.
I'm using GRBL with a vinyl cutter
and some times , with long travels , I've got some hard reset on the system.
Maybe it's the speed is too high
 250steps/mm
max speed 20000mm/mn
accel : 300mm/s/s

Maybe it's because my motors take too much of current (speed too high)

but I wonder if this is not a problem of pulse duration for the DC drivers
the default value in GRBL is 10ms
is a realiste value for the dcservo or it need adjustement?

thanks

misan, 2016-12-12:
That is correct. GRBL does a good job until the feedrate is more than it can handle. With 250steps/second a feedrate of 20000mm/min would require 20000/60*250 which is 83250 pulses/second while GRBL can only deliver 30000 pulse/second tops.

You only experience that problem on long travels because acceleration does not allow the max feedrate to be reach on shorter moves. 

It is  GRBL limitation. You can reduce microstepping if you absolutely need the max feedrate and have to use GRBL.

10 microseconds pulse on step should be ok.

Cdriko, 2016-12-13:
thanks
I set the max feedrate to 30000/250*60=7200mm/min and it works !

closed 2016-12-13 by Cdriko