misan/dcservo #18: Instructions for ESP

Wintertod, 2016-12-17:
Can't get it to work on an nodemcu module.
changed the pins to  #define, also  led pin does not work on all esp modules with internal led command.
Would suggest to program the encoder input by default with _PULLUP function as that's what most encoders work with.
also how does the server work? serial only promts the first lines and no communication afterwards possible.
General //dokumentation inside program could help much.

misan, 2016-12-17:
Hi, sorry you are getting trouble.

The code posted worked with my hardware. There is nothing wrong activating the input pullups if your encoder needs it. 

Server could be access with telnet or netcat program to port 23. 

I assume you are using dcservoESP_wifi.ino, right? 

Available commands can be retrieved with "h" command and listed in the usage section too. 

Are you ending your lines with carriage return or line feed?

Wintertod, 2016-12-17:
Yes , I'm using dcservoESP_wifi.ino.
Now I got access to my ESP, the  encoder works.
Thank you very much for your good work!

I'll list what I had to do, so other people have it easier.
nodeMCU "V3" with ESP-12E (works for every ESP-12E board)
1 Replace all "LED_BUILTIN" with "led"
2 Add "#define led 2" at the beginning or copy paste code below between const char* password and byte pos[1000];

// Opens a Telnet Server on Port specified below.
// specify the port to listen on
WiFiServer server(23); //Edit Port = Adress for telnet 192.168.xxx.xxx:23 |windows telnet won't work.
WiFiClient client;

#define M1 16             // D0 Output Motor 1
#define M2 5              // D1 Outout Motor 2
#define DIR 4             // D2 Input Direction
#define Step 14           // D5 Step input
#define encoder0PinB 12   // D6 Encoder B, white, edit pin to "pinMode(ender0PinA, INPUT_PULLUP);" for standard encoder
#define encoder0PinA 13   // D7 Encoder A, green, edit pin to INPUT_PULLUP for  standart encoder
#define PWM_MOT 15        // D8 for ESC driven Brushless Motor
#define led 2             // internal LED on ESP-12E on any other ESP use GPIO 1 (can be any pin you want)

closed 2016-12-17 by Wintertod

misan, 2016-12-17:
Great!

Thanks for the tips for other users!