misan/dcservo #19: not finishing sequence

mcarosh, 2017-03-19:
Hi Misan,
first of all, great work ;) 
But now i´m facing some strange behavior. My setup is a micro pro with 612fng and a faulhaber-dc-motor with quad-encoder. steps and DIR is sent from a Printrboard. So far I managed to get it working. My big problem now is, if I sent G0 X50, motor is driving to its position, but if I´m immediately sent G0 X0, the motor is not driving to position 50. It is only moving to ca. X14 and then turns back to X0. 
The same is happening when i open a file with these orders. 
Especially for milling I prefer to use equal sequences. 
So my question, am I doing something wrong or is there a limitation in your code or what is the problem. 
Maby you can help me.
But so far I can only say you are the man ;) who changed my life :) never thought to use my faulhabers :)
best regards
mcarosh

misan, 2017-03-19:
Hi mcarosh,

My guess is that the two commands are being joined by Marlin planner so a position error is happening. However such a big error (only reaching X14 instead of X50) seems too much for that explanation to hold. 

Alternatively, your dc-motors might be too slow to respond, so while still moving towards X50 the pulse sequence changes the reference faster and motor just changes its direction to follow the new target value (back to X0). This makes more sense to me if your motors have a significant reduction gearbox. 

So my guess is that if your motors are to slow to follow the commands you have your system configured wrong, asking your motion for an acceleration your motor cannot provide that places you in a position you report. 

Does it make sense to you?

misan

sanchosk, 2017-03-19:
This is a problem of feedrate.
Slow down the commands to the "steppers" and everything will be OK.
I had the same experience - not only the speed of the DC motor, but also the PID settings influence what will be the final allowed feedrate from the driving board.

mcarosh, 2017-03-19:
Hi Misan
unbelievable how fast you answer 8) 
and yes you re absolutally right. 
Just in this moment I saw my wrong speed settings in repetier. 
I´m so sorry for disturbing you on weekend and really really thx a lot for your great work again. 
I hope not to upset you again and hope you have nevertheless a calm weekend. 
thank you, thank you, thank you :)
greez 
mcarosh

closed 2017-03-19 by mcarosh