misan/dcservo #20: dc motor and servo  and motor shield adafruit !!!!! plzz need help!!/:

themomza, 2017-04-06:
i've been programming my robot with the adafruit v2 motor shield! everything works perfectly without the servo! but when i plug in the servo 
the DC motors of my robot stops working and the servo keeps turning from 0 to 180 degree. and when i unplug the servo the DC motors of my robots starts to work perfectly .
so my DC motors doesn't work when i plug in the servo!
can someone explain to me what the problem!

plzz i really need help!!??

misan, 2017-04-06:
Out of the blue, servos require high currents sometimes. If you power the
servo from Arduino's +5V supply the servo might bring it down ocassionally
freezing the processor in the process.

themomza, 2017-04-06:
Ohh i see! But im plugging the servo in the motor shield and the motor shield is plugged on arduino and im supplying it with 6 AA batteries ( the wires of the batteries are plugged in arduino and in the motor shield ) ..

themomza, 2017-04-06:
Im not using a vin jumper

themomza, 2017-04-06:
Do you think there is any solution?

misan, 2017-04-06:
what is powering the servo? +Vbatt? +5V?

themomza, 2017-04-06:
Can i email you a picture of my circuit?

themomza, 2017-04-06:
[image]

themomza, 2017-04-06:
As u can ser here i've tried the vin jumper and it didnt work still same problem... before i was plugging twi wires in the small blue thing (5-12vmotor) and it wasnt working

misan, 2017-04-06:
Please read carefully the instructions of your board
https://learn.adafruit.com/adafruit-motor-shield-v2-for-arduino/powering-motors

6V might not be the best choice either, read Arduino supply requirements
too.

Yours seem entirely a supply problem.

themomza, 2017-04-06:
Im using 9 V there is two more batteries that didnt appear in the photox.x ! But i think your right! It might be a supply problem! 
Thank youu  for your time btww ! :))