misan/dcservo #23: Trapezoidal speed motion pattern

kr4fty, 2017-05-14:
Hi misan, I hope you are well. First of all, I wanted to congratulate you on the excellent work you have done.

You could explain roughly the advantages of using the trapezoidal speed motion pattern.

Thank you very much and greetings from Argentina

misan, 2017-05-14:
In a nutshell, motion planning takes into account the physics of the motion
which are not account for by the PID (motor cannot accelerate faster than a
given maximum, so there is no way it would do so no matter how hard the PID
controller asks it to). So, motion is going to follow a given plan, where
velocity has a trapezoidal profile. This way position and speed is known at
any instant and controller can correct any deviation. Specially useful when
you want synchronized motion of several motors.

kr4fty, 2017-05-14:
Thanks for the answer and for your very good explanation