misan/dcservo #26: dcservo.ino assigns long to byte

DrStein99, 2017-08-06:
The "example" assigns "POS" as a byte, yet - assigns the LONG encoder position to this BYTE array....  Not sure how this actually would work

```
byte pos[1000]; int p=0;
volatile long encoder0Pos = 0;
double input=0, output=0, setpoint=0;

pos[p]=encoder0Pos; 
```

misan, 2017-08-06:
Thanks @DrStein99. 

It does not look good, but the initial aim was to use this feature only for PID tuning, where the tool would create a step from 0 to 100 (so all the expected values there would be with the byte range). 

But you are right that this feature won't work properly most of the time unless used under similar conditions.

Not using a larger data size was due to the scarcity of RAM memory on Arduino UNO, so for other platforms data type might be changed to integer or even long for reducing or removing the limitation you pointed out.

At any rate, it is good you mention it so now people can know about it as I failed to mention this.

DrStein99, 2017-08-07:
There are at least a couple other hang ups in the example I had to correct.

After I replaced the ENCODER translation to use the Encoder library, it solved ALOT.