misan/dcservo #27: Sudden reverse movement

ChristianLerche, 2017-08-10:
Hi Misan. 

Thanks for sharing your firmware with the world. 

I have a problem with the code. 
I have designed an extruder,tuned it using the tuning software, works like a charm. Commanding it from serial port works also. 
But, when controlling from driver board via dir step interface, the motor suddenly reverses the output, before stopping and extruding like normal. It doesnt go back to where it reversed from, just going forward like nothing happened. 
I checked position, target and PID output. Position and target were close together, all normal. PID was negative 255, so of course it was going backwards. 
I'm debugging right now, but perhaps ypu have seen issues like this before? Thanks.

misan, 2017-08-11:
-255 output means full speed backward. Is it possible your tunning was done without filament and now the parameters may not be right when the load changes?

I guess you check all wires to have a good connection.

If position and target are close together and yet PID is giving the maximum output that suggests too much proportional gain. But please note diagnosing this from afar is not an easy business :-D

ChristianLerche, 2017-08-12:
I guess I have found the flaw. It's in the encoder hardware. Pulses are not 50% dutycycle, and not 90 degree out of phase. Actually it awitches over on the same time at some point,so not really quadrature encoding. Somhow it has just gotten worse now. 
When I have fixed, I'll return here :)

misan, 2017-08-12:
Some encoders might require a couple of pull-up resistors on the output pins. If you fail to put them output signal may be a mess. Double check the specs of yours.

ChristianLerche, 2017-08-12:
Oh they're pulled up. Unfortunately the china designed PCB fails to mention that the hall-sensors are A3144 types, needing a voltage of at least 4.5V to function correctly. But the PCB states 3.3V. After removing the PWR LED and supplying the sensors with 5V, they're behaving better. Stil not quite quadrature encoding, but I intend to make my on set of magnets on the motorshaft. 

Btw, I repair/make/build/design electronics for a living, programming just comes across now and then ;) 

Thanks!