misan/dcservo #28: Motor voltage and homing question

paukstelis, 2017-08-13:
This is not specifically dcservo related, but it seems like there are some more general questions being asked here as well. Sorry if this is too off topic.

I'm using a scraped liquid handling robot (Tecan Freedom EVO75) to build a large 3D printer, primarily for the learning experience. Lots of reasons to not do it, so I'm doing it anyway! It original uses DC motors with closed loop servo control, though it uses custom motor drivers (the boards are also called DCSERVO). I've gotten the encoders working with dcservo, but I'm a bit stumped on the motor voltages. I don't have a way to trigger the motors from the original boards (that I have been able to figure out), but I can see that both Motor(+) and Motor(-) are at +10.7V. Any guesses if this would be the actual voltage at which the motors are run? The board is being fed 24V.

Second, how are most people handling homing with incremental encoders? Just end stops? Or is there a more clever way of finding position after a power cycle?

misan, 2017-08-14:
Hi,

I have no clue about the type of control the motors of robot can have.

Regarding end stops you can use regular ones or, alternatively, you could
use the loop information for determining when one end of the travel has
been reached. That functionality we could call "virtual end-stop" is not
included in the current code base as I am not sure everyone can see that as
a useful (and safe) feature.  The basic idea would be to slowly move
towards one end while monitoring encoder counts. once they increment no
more should be because the end of the travel has been reached, so the end
stop signal should be raised.

paukstelis, 2017-08-15:
Thanks, Miguel. I'm pretty sure I'm looking at 12V motors. 

I'd be interested in seeing an implementation of the virtual end-stop if you do have one laying around. I guess the safety issue would be a potential high current draw when the motor stalls?

sanchosk, 2017-08-15:
I believe there is one "virtual rnd-stop" implementation in my fork and one
pull request I've made in the past.

misan, 2017-08-15:
Hi Robert,

I reckon it is available just not for each processor available but only for
ATtiny. I reckon your pull request was merged.

paukstelis, 2017-08-26:
Thanks for all the help. Many more obstacles to overcome, but it running gcode. https://youtu.be/TTBmDFnIuko

misan, 2017-11-05:
Looking great.

closed 2017-11-05 by misan