misan/dcservo #29: Ad Stepper support

Wintertod, 2017-10-14:
Hello Misan,
Steppers are cheap and have much more torque than DC Motors in same size, can you add option to output Step/Dir signal?
I think it would be a great new feature.
Could you ad "bluepill" support? it's a great board with many features like hardware AB decoder and can bus.

misan, 2017-10-14:
Hi Wintertod,

You are right but steppers would need at least a couple of h-bridges to drive (one L298 would be enough) for those under 2A. But projects like Mecaduino already cover that solution space (plus they are based on an M0 I reckon). 

Outputting step/dir would assume there is an external driver but that would prevent a more thorough approach to motor control, like a vector drive, overcurrent and other strategies to deal with following error conditions. 

Bluepill support for dcservo should be simple if using STM32duino library though EEPROM would need to be emulated. 

At the moment I am working on other projects, but I might get back to some of these ideas in the future. Stay tuned.

Wintertod, 2017-10-14:
Thanks for the fast reply!
I already looked at almost all DIY closed loop control options and either the hardware alone costs a lot or they only support limited hardware that's not always available. 
Again I would like to see a controller only, this way you can take whichever encoder(AB/I2C 5$) you want and choose a Stepper-driver(8$) to your liking (Step/Dir input) coupled with either an arduino, stm32 or esp board(2.5$) you'll end up with a fairly cheap (15$) 5A driver solution with 9 to 12 bit resolution.
If I have to buy,or build myself, Mechaduino, smartstepper, Tarocco, Vesc(supports BLDC), stmbl would just be too expensive(40$ to 100$ + motor) when I can buy a JMC 0.6Nm ac servo or 3Nm closed loop stepper with integrated driver each 80$.
I think it is a shame that so many makers take so much effort to write code and design hardware only for it to be more expensive than industrial solutions in the end. 
And you don't need vector control and such, just a good PID. I tried to control a stepper with as5600 and pwm input like an RC servo, I got the positioning working but till the code really works it will take much too long, I'm just an hardware guy. 
I think what we need today is easy-to-interact-with cheap precise motors with decent torque so makers around the world can build new and innovative projects without having to be a jack of all trades or having a big budget. Just saying that all that open source won't be of any use if you can't get it cheaply.
I have seen you using bought closed loop motors on your YouTube, and with your reply I think this project is dead, but thank you for reading this long mail and I hope seeing some useful cheap diy closed loop systems in the future so I will stay tuned and continue to try working on my "rc servo stepper motor"
Have a nice evening and thank you for your work so far :D

misan, 2017-10-14:
I do agree, it is difficult to keep the cost low and easily the off-the-shelf products from Leadshine or JMC will be easier and cheaper than a pure DIY approach. In fact, that is what I am using on some DIY CNC machines I built.

The problem with closed-loop steppers is what to do when the stepper stalls. If you do the simple thing of keeping sending pulses to the drivers there is no guarantee that it will work at all. Depending on your application, a certain following error may not be later compensated for. But if you keep the driver within the loop you can try doubling or tripling the coils current to get momentarily extra juice from the motor so as to avoid the stall condition. So it gets complicated if you want to get decent performance and with that the drive gets more expensive, reaching easily the cost of a commercial unit.

DrStein99, 2019-02-02:
Am I missing something here?  It's just a pwm-out & high/low step signal.  The direction is high-low toggle.  The step-motor driver handles all the current and torque processing.  If the stepper driver (like TMC2130) has a stall-out pin, it's only a few lines to catch the error and halt the code.  

The encoder input on a motor with a stepper driver attached to this project makes this an actual closed loop control.  On a professional grade system is elaborate with many encoders on the motor, gearbox, and linear all monitoring motion to catch any variance and prevent against part and machine damage.  1 motor, 1 encoder is the basic foundation for a closed loop system.

misan, 2019-02-02:
@DrStein99 I have not read all these documents: https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/tmc2130-driver-infos-and-modifications-t15835.html but my understanding is skipped step detection is not a pin but a more convoluted process of reading register variables and using properly adjusted thresholds. I am afraid it does not catch all missing step conditions.

DrStein99, 2019-02-02:
@misan There list of reasons for steps lost could be as long as the datasheet for the TMC2130 (which is about a ream of paper).  The TMC2130 is just the end motor amplifier in the control process.  My understanding Trinamatic, they make their own control chips and evaluation boards that handle all the acceleration curve processing, motor feedback and encoder signals needed to accurately drive all 3 types of stepper, DC or BLDC motors.  Those evaluation boards are not cheap to buy & try.  Not really an option for spending $160 on the entire 3d printing gantry system with control unit. 

This PID loop application you have made with the with the recent addition to the acceleration curve is an important piece to the whole puzzle.  I run this on a 32-bit STM32F103 which has more than enough resources to process at least 1 axis.  We can use this for any motor/motion purpose, without spending the $300, $1,300 or $3,200 for a Trinamatic motion control evaluation board.  Of course they actually offer the PC software to interface with it to tune and configure, which is really important - but not necessary for someone who specifically knows what they are looking for and what they are doing.

misan, 2019-02-02:
Yep,  I saw their FOC brushless driver kit and its price and it did not fit
the price point I expect when giving a try a technology for mere
curiosity.

I reckon ST32F103 can happily handle more than two dc motors with
incremental encoders.

But I can see manufacturers really love steppers as almost none (I actually
know two) seem to be using closed loop dc or BLDC motors that would bring a
safer printing experience IMHO. That is why I am obsessed with keeping it
simple and cheap so it can eventually be accepted.