misan/dcservo #31: Adding a hard endstop detection

pppalain, 2017-11-05:
I am designing a 3d printer using these motors.  I have no intention to add endstops.  Here is an option I will be adding.

If motor is energized and the counter can no longer count because of a physical barrier.  The controller can detect that to signal an endstop.  This should reduce physical switches for each axis.

When an endstop request signal is detected motor should only be energized enough for motion.

proposed autodetect the minimum pwm needed for motion upon startup
proposed endstop request input pin. from an external controller such as Marlin
proposed endstop detect output pin. To controller such as marlin

misan, 2017-11-05:
No endstops is what the new Prusa i3 Mk3 is doing too. Simpler and easier to use.

I can think of two different approaches:

1) there is a special motion command for "homing" where motion towards one end happens at  low speed while error signal is monitored. once error starts increasing (no more encoder counts) motion stops and location is meant to be the new 0 location. The event maybe signaled by raising an output pin

2) any movement is checking whether or not there is suddenly a block of the encoder count. If that happens motor is de-energized and an exception is signaled (maybe to both serial port and as an output pin)

The problem with the second approach is that some printing errors might trigger that condition though some kind of protection may always be welcome. But initially, I would favor the first choice. But that begs the need of a way of setting the controller into the "homing mode" which can be done with a serial command or an additional output pin ...

sanchosk, 2017-11-05:
I made a bit different approach in my ATtiny85 code. On power on, I increase the power slowly and monitor, if the encoder is moving. If it does, I am sure I am not at the end. By the change of the encoder I realize if the current direction of motor is positive or negative..
If it did not move, I try to move to the other direction - perhaps I was at the end already. If it moves and the encored registers movement, I, again, gain the direction.
Next step - increase the power by some percentage over the minimal power and start to move in negative direction while monitoring the encoder. Once the encoder stops moving and the supplied power exceeds predefined threshold, I know I am at the end. This position is now stored as 0 and that's it.
The same procedure now can be used to get the max endpoint, but I did not found it very useful, so I did not implemented it.
Check the code for ATtiny85, you can simply copy it into your own code.

misan, 2017-11-05:
That is a good point, instead of using a pin, just do it after controller reset (at power on). So you can set the output pin anytime controller is at 0 location.

misan, 2017-11-05:
However, I can see that this feature may be nice for some users but it will need to be somehow conditional so those who do not want it can disable it easily.

sanchosk, 2017-11-05:
Due to the scares amount of pins on ATtiny85 I used the EN pin from the mainboard to switch the power for the ATtiny chip. I set the firmware to keep the EN always on until I issued M84. After M84 I triggered G28 X and voila ;)

pppalain, 2017-11-18:
I have completed my implementation and will share shortly.  At this time it seeks a negative endstop. see: https://youtu.be/kTUaKS-4Spc

michaeljball, 2017-11-18:
This is definitely the way to go!  
As well as monitoring for halted encoder count, I've changed my motor drivers to the NXP 33887 which monitors current (load) and provides feedback to an ADC. This can used for a number of things, but endstop detection was my primary use case.

You can also use the current feedback in the acceleration PID, but I've not perfected that yet.

pppalain, 2017-11-18:
in setup the indicator pin is set to output
help(); is removed
then homing();  is called in the end
so in this version homing is triggered upon reset.

void setup() { 
  pinMode(encoder0PinA, INPUT); 
  pinMode(encoder0PinB, INPUT);  
  pinMode(ENDSTOP, OUTPUT);  
  pciSetup(encoder0PinB);
  attachInterrupt(0, encoderInt, CHANGE);  // encoder pin on interrupt 0 - pin 2
  attachInterrupt(1, countStep      , RISING);  // step  input on interrupt 1 - pin 3
  TCCR1B = TCCR1B & 0b11111000 | 1; // set 31Kh PWM
  Serial.begin (115200);
  recoverPIDfromEEPROM();
  //Setup the pid 
  myPID.SetMode(AUTOMATIC);
  myPID.SetSampleTime(1);
  myPID.SetOutputLimits(-255,255);
  homing(); // 
} 

/*  enables to detect obstacles such as hard stops or soft rubber stops 
 *  without the use of a limit switch since it looks at the growing error 
 *  when such an event occurs.  
 *  an output pin is asserted after stop is detected for 2 seconds
 *  
 *  
 */
void homing(){
  long tstamp;    
  int error=0;
  int scanning_steps=15; // how fast you want to scan for home
  int max_error;
  float homing_power=0.6;  // power multiplicator for homing. suggested between 0.1 for 10% and 1 for 100%
  max_error=(scanning_steps*7+1);
  digitalWrite(ENDSTOP,0);  // Turn external pin low
  Serial.println("homing ...");
    while( error>-max_error){   // loop while error is less than max_error an obstacle or rubber stopper will make the error increase at each interval
      if(millis()-tstamp>6) // decreasetarget at desired time interval (6 default) 
        {
          setpoint-=scanning_steps;   //decrease target
          tstamp=millis();  //stamp the time
          Serial.print("setpoint ");          Serial.print(setpoint);
          Serial.print(" encoder ");          Serial.print(input);
          Serial.print(" error ");          Serial.println(error);
        }
    input = encoder0Pos;
    error=setpoint-input;  
    while(!myPID.Compute()); // wait till PID is actually computed
    pwmOut(output*homing_power); 
  }
  encoder0Pos=-70; // detected limit is now -70 to (if a soft limit is set like rubber motor would always try to push if zer0)
  target1=0;
  digitalWrite(ENDSTOP,1); // assert output pin for 2 seconds
  help(); // display help while waiting for the pin assertion
  delay(2000);
  digitalWrite(ENDSTOP,0);
  
}

misan, 2017-11-18:
Great job and thanks for the video and code.

closed 2017-11-18 by misan

michaeljball, 2017-11-18:
This concept is also good for dynamic bed leveling with just the extruder nozzle
When the Z-Axis stalls, you've got your Zero point.

misan, 2017-11-18:
But output pin seems odd for a 3d printer firmware. I would put that on the
main loop() so

if(encoder0Pos<0) digitalWrite(ENDSTOP,1);  else digitalWrite(ENDSTOP,0);

That would feel more natural to the 3d printer firmware as the motor is
reacting to the steps and direction signals when homing ...

pppalain, 2017-11-19:
the problem with this is that a digitalWrite takes a long time and may affect performance.  I will look for another way which is not so greedy.

sanchosk, 2017-11-19:
Perhaps I would suggest to take inspiration from Repetier 3D printer
firmware and their way how to do fast digital writes using registers...

misan, 2017-11-19:
In fact, I am already using PINx macros with a binary mask to get faster reads on encoder interrupt. I gather same could be used to speed-up output, though if using my proposal, the loop () latency is 1ms which is much slower that digitalWrite.

misan, 2017-11-19:
Food for thought http://forum.arduino.cc/index.php/topic,4324.0.html

pppalain, 2017-11-19:
as I understand it, using digitalWrite is compatible on many platforms.  not so with bitwise register manipulation.  I have solved the problem by only issuing a digital write during the relevant changes.  normal operation will be largely unaffected.

misan, 2017-11-19:
Yeah, you are right, masks and port-wide operations are more difficult to
follow and board-dependent. Plus any time used here won't delay the next
output of the loop as PID compute waits till the next millisecond so
pwmOut should always be delivered on time.

paukstelis, 2018-02-01:
As suggested by michaeljball, I'm trying to get this to work for bed leveling using the nozzle as a probe in Marlin. Not having much luck, however. Triggering the endstop isn't the issue, but every point in a mesh leveling (UBL) increments by some negative value. Presumably this is because Marlin is calculating the z position as decreasing even as the Pro micro (in my case) is detecting a stalled motor/encoder.