misan/dcservo #37: two motors one nano

pppalain, 2017-11-21:
I plan to undertake a two motor / encoder  version of the software.  Let me know if there are technical issues or if someone tried and failed.

I intend to use the pin change interrupts and the https://github.com/GreyGnome/EnableInterrupt library.

misan, 2017-11-21:
If you use two pins of the same port for an encoder it is ok as they share the same interrupt code. 

The main problem is motor max-speed is reduced as now as the Arduino has to deal with double the interrupt rate. 

ESP8266 is a much faster processor that can handle much higher interrupt rate and you can see the ported code of dcservo too (in case you get trouble with Arduino UNO).

pppalain, 2017-11-22:
My target destination is for a 3d printer.  it is rare if at all that both motors would operate at the full rate of my encoders simultaneously.  **I don't even know if Marlin can step that fast.**  

I am using a 180 Line per inch printer encoder which multiplied by 4 gives me 720 readings per inch or about 34 microns; about 11700 for 400 mm.  at top speed my head moves at about 1 to 1.5m/second without skipping readings.  This is much better than the top 300mm/s for a stepper motor.  I think the reason I was skipping earlier was the band being not adequately mounted on the assembly.

misan, 2017-11-22:
Then I do not think you will run into trouble. Some people use very high-speed motors and/or high-resolution encoders and the Arduino starts getting trouble above 60k interrupts per second or so.

pppalain, 2017-11-23:
I have now rewritten the ISRs for both encoders and I have rewritten the eeprom save to be able to save the parameters for the extra motor.  So this is comming along well.

misan, 2017-11-23:
Well done. I have been failing again with the PID Autotune. While PID
library works well (there is version 1.2 now) I have not yet managed to get
AutoTune to work (well now it is working but I get ridiculously low Kp
values so it is not useful).

Please note that for stepper emulation, interrupts come from two kinds of
sources: encoder pins and step inputs. Marlin can reach 40Khz step
frequency that times two (as you have two motors) will directly exceed what
your Arduino can handle. But on top of that you also have the contribution
of the encoder pulses, so I am not sure the experience will be completely
trouble-free.

At any rate, please keep me posted on your progress.

pppalain, 2017-11-26:
I ran into a bit of a problem trying to use PWM output 5-6 for my second motor.  I was unable to change the PWM frequency without affecting a bunch of stuff I think PID uses Timer 0.  So... I used pin 9-10 but one PWM pin for each motor and another digital IO pin for direction.

It appears to be working well.  

I am working on a pretty much complete rewrite.  So you may not recognize much in the end.

pppalain, 2017-11-27:
** reminder ** 
when bit masking, PORTB&=0xF7 
not the same PORTB=0xF7;
writing it correctly the first time could save a lot of time.
Just lost over an hour due to that.

misan, 2017-11-27:
I have seen the L298 Arduino shield from DFRobot does need only one PWM input for Enable and one digital input for H-bridge direction. That may reduce your pin count needs.

misan, 2017-11-27:
definitely &= != =

= instead of == use to waste my time too :-D

pppalain, 2017-12-12:
I have designed a few PCB for this and they have been sent to production.  It is a shield for the Nano having the L298N on it but with connections for the encoders as well.  one board plugs into the ramps stepper motor slots x and y to extract 5V and GND and x step x dir ystep ydir.  I understand that this is very specific so it may not appeal to all.  I am considering releasing the PCB drawings and schematic as a branch or a new git project.

pppalain, 2017-12-12:
PCB design was made in KiCad

pppalain, 2017-12-12:
here is the schematic diagram - mostly interconnections, nothing very complicated
[motorPIDL298.pdf](https://github.com/misan/dcservo/files/1552004/motorPIDL298.pdf)

pppalain, 2017-12-12:
here is the PCB the main PCB is 63mm x 71mm
[motorPIDL298-brd.pdf](https://github.com/misan/dcservo/files/1552339/motorPIDL298-brd.pdf)

misan, 2017-12-12:
Sounds like a compact solution.

I'll be happy linking it on the README in case you want me to.

pppalain, 2017-12-14:
I already made my first board obsolete.  It is not even back from production. This one is using the A4950 3.5A 40V  o it's good for a 120W motor WOW.  Instead of using the PWM i decided to use the VREF to change the current.  Changing the current changes the torque proportionally.  I use the I2C MCP4725 as a 12bit D/A convertor.   It will be possible to have a 12 bit resolution torque control instead of a 8bit pwm.  The idea is to set the RS as close to the motor current as possible to the motor specifications and then vary the VREF with the D/A convertor.
[motorPIDA4950.pdf](https://github.com/misan/dcservo/files/1559713/motorPIDA4950.pdf)

misan, 2017-12-14:
I did not know that H-bridge but specs are very nice. I reckon you can use 16-bit PWM on the 328, but I am not sure the difference it can make. Using a D/A and torque control seems to be a big improvement. Cool!

bogus105, 2018-02-10:
Any progress guys? I'm so curious about it:)

pppalain, 2018-02-10:
I received the PC boards but I am still waiting for a few parts.  I will send pictures of what I have.

misan, 2018-02-10:
I did try A4950 and I really liked it but somehow I got sidetracked
http://fightpc.blogspot.com.es/2018/01/more-samd21-m0-weirdness.html

I used that protoboard to solder the SOT8 part and give it a try.

pppalain, 2018-02-10:
[image]
[image]
[image]

Here is what I have so far

bogus105, 2018-03-02:
This is nice. How about torque control. Can you write more about that? By the way: do you guys know any h-bridge IC suitable for motors rated at 48VDC or even higher: 60-90 VDC? Got some sweet electrocraft motors (they are 48V and 60V) and i'd love to see them running at their full specs. 
So far used L298:
https://www.youtube.com/watch?v=vHufLMh4xEI&t=18s

as well as BTS7960:
https://www.youtube.com/watch?v=icBFc65dfvA

But got some troubles (PID with tuning i believe) controlling larger motor on BTS7960 - too long time it takes to reach position.

misan, 2018-03-02:
Hi bogus105,

For torque control current measurement is needed, so that will need to be taken into account for H-bridge selection. Instead of closing the loop with the position, it is the target current what we control (still by changing the PWM output). 

Some advanced controllers have a torque controlled inner loop surrounded by an outer position or speed control loop.

I have not explored higher voltage than 40V H-bridges. I like the 3.5A/40V A4950. 

Larger motors may have more inertia, but remember top speed is usually not what you need for a position control as your motion controller will be set not to ask speeds any faster than what your motor can achieve.

I reckon for 60V or higher you might need to build an H-bridge with gate-drivers and discrete components.

misan, 2018-03-02:
I just saw this 1.5A 60V bridge https://www.mouser.com/ds/2/294/NJM2670_E-56778.pdf

happytm, 2018-03-24:
pppalain & bogus105,

Any progress made on your projects ?

bogus105, 2018-03-24:
not yet. Misan you said i might need to make my own h-bridge instead of all-in-one-chip h-brigdes. Actually i have 4 pieces of UHU controllers (anyone remember UHUs for a beer?:)) so i already have a great h-bridges. Just need to throw microcontroller chip and hook up PWMs to it. 
oh, I forget about  oe m ore thing - i need to find time for it...

misan, 2018-03-25:
UHU controller does the same job as dcservo
https://www.youtube.com/watch?v=PBFw8pMfcck and if yours includes the
H-bridge (I reckon it was sold as a single chip too) then you should be all
set. That is for the hardware side, for the "I need time" side I am afraid
I cannot help :-)

pppalain, 2018-03-25:
I have purchased a large CNC machine in December.  At the moment, I am putting all my free time to optimize it.  still needed on my CNC: PWM control of spindle, Automatic tool change.

I did not have time to program / troubleshoot the new boards.  I will keep everyone updated.

happytm, 2018-03-28:
Thanks pppalain.

Please keep us posted.

iramuhammad, 2018-04-26:
What happened with the 2 motors per Nano project  😊☺

misan, 2018-04-26:
It can be done, but it will only handle well slow motors and/or low-res
encoders. Sample code on youmagine.

pppalain, 2018-04-26:
I have not tested the board yet.  Have not had the time to make a test jig and sit down.  It is not abandoned.  Just one of those things.

misan, 2018-04-26:
Life happens ;-)

sanchosk, 2018-04-26:
For me it's not relevant - ESP8266 is cheaper, way faster and able to use
the same Arduino libraries (at least the ones I've needed).
I also stopped using ATtiny85, although I have line 12 around still...
Life goes on.

pppalain, 2018-04-26:
I was using this  180 LPI encoder strip [https://www.aliexpress.com/item/infiniti-FY-3208R-inkjet-printer-encoder-strip-180dpi-4-5m-1-5cm/32584547187.html?spm=a2g0s.9042311.0.0.cAqkK3]

and when multiplied by 4 gives 720 transitions per inch or 35.28 micron  perfect for a 3d printer which is my goal.    I was able to move on one motor 1 m/s  could have gone faster but the 400 mm printer axis are too small for that.  at that speed, the belts are all stretched and the whole assembly wants to self destruct when doing a stability test.

concifederico, 2020-01-17:
Hi, still looking for the dual code! I would like to move 2 encoders in sync mode for the Y axis plus 1 encoder for the X one, slow speed. Would be nice to avoid starting this from just dcservo.. regards!

happytm, 2020-06-15:
@misan  I have ordered some A4950 H bridges. I know you tested it a while ago. I wanted to try it with 24V  dc motors which have built in optical encoders. What would be your suggestion for micro controller. I would like to use ESP8266 or ESP32 for built in wireless control. I also have some 328p and STM blue pill micro controllers if you think they are better suited.  I will use 2 of these motors for moving car platform with 1 free pivot wheel in front. 

Thanks.

paukstelis, 2020-06-15:
I found Blue Pill being very easy to use with quadrature encoders. You can
use the hardware timers directly which means little to no software overhead.

happytm, 2020-06-15:
@paukstelis Could you please share any wiring scheme with blue pill and code you might have.

Thanks.

happytm, 2020-06-15:
Thank you for the link to the code. What about wireless control of the moving platform? I Still have to use RF hardware (like NRF24) right?

That is why I was eager to use ESP8266 or ESP32 for simplicity in hardware.

I would seek someone's  feedback about A4950 using with  WiFi micro controllers. What are the disadvantages of using WiFi micros?

Thanks.

bogus105, 2020-06-16:
I was considering adapting Misan code for ESP8266 however ESPis not very good for that. PWM is limited only to 1kHz (if i remember right) and not many GPIO. ESP is fast, much faster than Atmels, however, as it is two cores, it is not "as real time system" as the old Atmels. However:) Mose of the servos for DIY doesn't need 10000 ppr encoders on the shaft and 20 parameters to be set in PID. 
So we can try to draft some requirements to implement servo control system based on ESP8266 (or ESP32) . IMHO we will need to use external peripherals like specialised PWM chip over I2C (like PCA9685 16 channel 12 bit PWM), and some quadrature encoders decoder to do the time sensitive job.  Then ESP will just communicate with peripherals to collect data on position and set PWM values at set intervals. What about that?

happytm, 2020-06-16:
This could be the good solution.

Thanks.

misan, 2020-06-17:
@happytm ESP32 does have hardware PWM but ESP8266 does not. A4950 works quite nicely. The limiting factor for ESP32 is the interrupt latency, which will limit motor max speed. https://esp32.com/viewtopic.php?t=422

@bogus105 EPS8266 is an excellent choice for 2.4G wifi communication but it sucks at real-time tasks. On-chip PWM is one of these real-time tasks, so it cannot be trusted. Using additional chips for doing other tasks (such as PWM) is one way to address that shortcoming but as you add more components you quickly create a solution that is bulkier and more expensive than using an STM32 that can handle real-time tasks and that has a set of built-in peripherals that offload most of the time-sensitive tasks from the CPU (as quadrature encoder input for timers and timer-based PWM generation). It might be cheaper if wifi is a must to join together an ESP8266 and an STM32. Now with black pills featuring '401 and '411 you can much faster cortex-M4 processors with built-in FPU for less than $5. 

What you lose going this route is the simplicity of having to create code for only one processor. However, if you use ESP-link firmware for the ESP8266 you can focus on the STM code only.

happytm, 2020-06-17:
@misan  Thank you for your feedback. I would first start with  blue pill & esp8266 with ESP-link as you suggested because I have both in stock with me. As I never used blue pill for this purpose can you please tell what pins should I connect with motor & quadrature encoder input. It is nice to hear 
that A4950 works nicely. I have 40 motors with encoders (I got a lot for $25) and I wanted to find solution for it. I don't have specific use cases beside using it as either linear actuator or robots so precision like CNC or laser printer is not absolutely necessary. I will be just tinkering with it.

Thanks.

happytm, 2020-06-17:
While searching for suitable micro controller I came across this repo. It makes this low cost chip arduino compatible. I think it definitely looks better than Attiny85 and may be as good as nano for motor control purpose. The pros are it has built in USB and it has up to 6 capacitative touch button inputs. Repo is brand new but it claims to make programming this chip easy. Repo is link below:

https://github.com/DeqingSun/ch55xduino

for hardware design  I found this:

https://hackaday.io/project/169671-ch552-dragon

Around 30 cents each and available at LCSC which has PCB and assembly service for cheap we could have inexpensive and complete solution for motor control.They also carry A4950 for around 60 cents.

https://lcsc.com/search?q=ch552

What do you think? Please discuss.

Thanks.

happytm, 2020-06-17:
Following older video but shows small footprint hardware with 2 touch buttons: 

https://www.youtube.com/watch?v=EMDJe7JQ76g

Thanks.

ithinkido, 2020-06-20:
>  You have to use the specific timer pins defined in the quadratureBluePill.h (which I did not write).
>
@paukstelis  Sorry, I am still not sure I understand how you have connected your encoders. I see 3 pins defined for 3 encoders , dont we need 2 per encoder ( ie 6 total) or have I missed something ?

paukstelis, 2020-06-20:
> > You have to use the specific timer pins defined in the quadratureBluePill.h (which I did not write).
> 
> @paukstelis Sorry, I am still not sure I understand how you have connected your encoders. I see 3 pins defined for 3 encoders , dont we need 2 per encoder ( ie 6 total) or have I missed something ?

See the end of that header file. Each encoder is on a timer, you can use timers 1, 3, and 4 so this shows you which is channel A and B for each timer:
```
byte pinA[5] = {0, PA8, 0, PA6, PB6};
byte pinB[5] = {0, PA9, 0, PA7, PB7};
```
Where the array number is indexed on the timer number.

paukstelis, 2020-06-20:
So to put simply PA8 and PA9 for one encoder, PA6 and PA7 for the second, PB6 and PB7 for the third.

ithinkido, 2020-06-20:
Thanks, I had not seen that. One more question please, one of the encoders has been commented out here
https://github.com/paukstelis/dcservo/blob/e0c44d8f1fdccdd1aaf51a02abd32566ae486bc2/dcservostm32/dcservostm32.ino#L66
Is that for a particular reason?

paukstelis, 2020-06-20:
> Thanks, I had not seen that. One more question please, one of the encoders has been commented out here
> https://github.com/paukstelis/dcservo/blob/e0c44d8f1fdccdd1aaf51a02abd32566ae486bc2/dcservostm32/dcservostm32.ino#L66
> Is that for a particular reason?

Because I am not very smart! I poorly selected my pin/timer combinations when designing my PCB, so the motor driver PWM ended up on the same timer channel as my Y encoder. You can't have them both.

ithinkido, 2020-06-20:
Which pins would you recommend to use?

paukstelis, 2020-06-20:
Honestly, it has been a long time since I worked on that project, but you could put motor driver pwms on timer 2 (PA0, etc.).

ithinkido, 2020-06-20:
Perfect, thanks for all your help

happytm, 2020-06-20:
@ithinkido I am curiously following your conversation. If you find good pin out solution for STM please keep us posted here.

Thanks.

happytm, 2020-12-06:
@ithinkido @paukstelis  Can either of you publish diagram (hand drawn OK too) like on main page of this repo with bluepill.

How many motors are controllable with one blupill device using motors with optical encoders ?

Thanks.

paukstelis, 2020-12-07:
You can have 3 I believe. Timers 1, 3, and 4. This assumes an open collector type (2 outputs/encoder).
You can find the little library here that describes the pins: https://github.com/chrisalbertson/quadratureBluePill
Wiring depends on your encoder. Just keep in mind if you want to use PWMs for anything else those must then be on Timer 2.

misan, 2020-12-07:
@paukstelis That is brilliant, three motors controlled by a single bluepill. IIRC, timer 2 could create up to three PWM signals.

happytm, 2020-12-07:
Thanks  @paukstelis .

paukstelis, 2020-12-07:
> However, the use of Timer1 means USB port cannot be used for comms, leaving only left USART3 for that purpose.
Hmm...I need to check that, because I just set up a project that has an encoder on Timer 1 and I have been communicated with USB. I haven't noticed any issues at all.

misan, 2020-12-07:
@paukstelis I gathered that insight from STM32CubeMX app for STM32F103CB chip.
[image]

[4 more replies]