misan/dcservo #38: Bad implementation of PWM overheating motor

pppalain, 2017-12-04:
I revisited the implementation of PWM after burning a motor (melting the 3d printed carriage) and getting a really bad electronic burning smell.  It shouldn't have gotten that hot.
I am using the l298 and I noticed this from the datasheet:

if the motor is enabled and the outputs are both high or both low, the motor is in braking mode.
if the motor is disabled the motor coast.
I had in the past experimented with this but forgot about it.
The PWM pin should be the motor enable.
dynamic breaking should only be activated when error is 0.
otherwise the motor should be coasting.

so 3 pins are needed for correct operation.  One PWM pin and two direction pins

The current implementations make the motor turn a bit and immediately break.  
dynamic breaking makes the motor heat significantly.  This is done at every pwm cycle (31khz)

This is something I had done a while ago but completely forgot.

misan, 2017-12-04:
Hi Alain,

Sorry to hear that.

I am aware of that "feature" with the bridge in slow decay mode (
https://www.precisionmicrodrives.com/tech-blog/2017/02/26/decay-modes-motor-h-bridge-drivers
).

Depending on the model of H-bridge enable pin is available or missing so I
opted for a solution that in fact shorts the motor wires (braking action)
during the OFF periods of the PWM.

If the bridge is disabled the bridge diodes will conduct the coil current
back to the power supply. I guess that is going to be softer on the motor
and will return some energy back to the supply instead of turning it all to
heat. Please let me know if you see it works better.

Thanks a lot,

Miguel

pppalain, 2017-12-04:
Oh you don't have to be sorry... I have thousands of motors and lots of smoke left.  I think different drivers implement things differently.  The TB6612 calls it a short brake.  it happens whenever the PWM pin is low for this chip.
 
The reason I am so concerned is that some systems may have run the motors hot and abandoned.  I am modifying my implementation for my 1 nano 2 servo which worked well.... before the destruction of the motor.  1 m/s for a small motor.... I would like to see a stepper motor do that.
The heating happens not when the pwm is at 255 but it happens when the motors are tracking a pulse.... so always a percentage on and the rest of the time braking.