misan/dcservo #39: problem with my project.

suradeer1688, 2017-12-26:
Hello, i am new guy to write program. I am doing the project about the Robot goalkeeper. First, i would like to know if  we use this code to make a closed loop dc motor, is it possible? second, I have incremental rotary encoder but i do not know how to change from potentiomenter to incremental encoder. Third, how abot the STEP and ENDSTOP. I use L298N and there was the 2 input and pwm channel for 1 motor.
 Thank you

misan, 2017-12-27:
Hi,

1) Yes, you can use this project to get closed loop control of a dc motor
using an L298N.
2) My code uses an incremental encoder so you do not need to change that.
3) My project allows an external controller to set two signals STEP and
DIRECTION to control the reference position of the motor.  We recently have
added a new feature for the control to do an automatic home probing, so it
can detect the end of the travel on one direction and use that to act as a
virtual end-stop, not sure this feature is useful for you, but it is
definitely optional.

I would love to see some pictures/video of your system once you have it
working.

Best regards,

misan

suradeer1688, 2017-12-27:
Thank you so much for your information. I just start to do this project and now I'm on research. If the project is move on, i will get some pictures to you. But the information as you mentioned, which code that you do automatic home probing? is it dcservo file? Do i need to connect wire STEP and ENDSTOP?

misan, 2017-12-27:
ENDSTOP is an output you can ignore but STEP and DIR are input pins you can use or set to HIGH or LOW if you do now use them.

suradeer1688, 2017-12-27:
OK thank you