misan/dcservo #40: Do i still need a stepper driver?

pbbob, 2018-01-02:
Hello.I am designing a 3d printer with 3d printed parts and scavenged parts from a bunch of 2d inkjet printers. I have found that 2d printers use dc motors with optical encoder disks, instead of stepper motors like most 3d printers. I was looking for a way to control the printer, and I think dcservo is a good option. I am trying to learn as I progress in the project, and I have a question about using dcservo.
to control the printer, I am going to use an arduino mega, with modified [marlin firmware](https://github.com/MarlinFirmware/Marlin), which is the controller for most [reprap](http://www.reprap.org) printers. Printers that use stepper motors use an a4988 stepper driver chip. I am planning on using 4 attiny85s to control 4 dc motors.(one for each axis and 1 for the extruder). Do i still need an a4988 (or similar) stepper motor driver, or can i just hook the attinys directly to the arduino mega?

pbbob, 2018-01-02:
Also, I have been having trouble finding answers to some other things. This seems like a good place to ask, so if i have any future questions, can i ask them here?

sanchosk, 2018-01-02:
You don't need stepper drivers, but you will need a motor drivers, still. I
used multiple, my worst results were with L298, best with one I don't
remember, but I can try to find it tomorrow. I predict you will face
difficulties with servo-driven extruder, the rest of axis is piece of cake
- I've made it out of 2 printers long ago - linear strips for x and y and
circular for z screw.
I might be able to gather an old youtube video of it somewhere...

misan, 2018-01-02:
I do agree with Robert: extruder needs to use a geared down dc motor for good results.

A new feature you may one to use is the "virtual" endstops, that will save you some wiring and components while still providing you the endstop signal for each axis.

Each motor controller is linked to the Mega to *-DIR and *-STEP pins (replace * by each axis name).

sanchosk, 2018-01-02:
Here is the very old video...
https://youtu.be/xI1BPWRbdvo

Kadah, 2018-01-04:
I used the print head motion assembly from a salvaged inject for the x axes of my first printer using this project to control the DC motor on that.

It worked worked, but there were some inherent issues with the setup. Backlash was rather bad, and there was a minimum movement amount for motion. So squares would be slightly trapezoidal, and circles came out tilted ellipses. As long the corners of a print weren't required to be high tolerance. I got enough usable parts out of the setup to slowly upgrade my Repstrap and improve print quality greatly.

misan, 2018-01-04:
@Kadah if motors are underpowered (or a too low gain is used) you can see this behavior. Remember just raising the h-bridge power supply voltage is giving any motors an extra kick (at the expense of more heat).

sanchosk, 2018-01-04:
Exactly my experience - moving from 12 to 24V did a lot of difference un my
case. But the drivers I liked supported only 12V :(
The newer ones were, on the other hand, too expensive for all 4 axes.
So I scrapped the project in the end.

misan, 2018-01-04:
I am waiting for a batch of inexpensive A4950 3-5A/40V 8-pin SO h-bridges
from China.  I've got 10 for $6.

sanchosk, 2018-01-04:
Wow, that would be something - most of HP printers work with 36V power
supply and this would greatly impact the performance!
Looking forward to your review (hopefully with video :D).

pbbob, 2018-01-04:
@sanchosk when you said your worst results were with the l298, what did you mean? what was worse about them? also, do you know which one you had best results with?

sanchosk, 2018-01-04:
High internal resistance = small current for the motors = small torque
Also, there was aome resonance sounds
And it was heating a lot
Just to name a few...

pbbob, 2018-01-04:
@sanchosk what h bridge would you reccomend?

sanchosk, 2018-01-04:
I am not the expert, unfortunately. My good experience was with TB6612 (
https://www.aliexpress.com/item/Free-Shipping-5pcs-lot-Dual-Motor-Driver-1A-TB6612FNG-for-Arduino-Microcontroller-Better-than-L298N/2027582523.html)
but for a bit bigger motors the 12V was not enough...

pbbob, 2018-01-04:
@sanchosk your link is broken

pbbob, 2018-01-04:
@sanchosk what do you mean that the 12v was not enough?

sanchosk, 2018-01-04:
Let's try this one:
https://www.aliexpress.com/item/Dual-Motor-Driver-1A-TB6612FNG-Microcontroller-Better-than-L298N/32465698640.html

pbbob, 2018-01-04:
ok. that works. thanks

sanchosk, 2018-01-04:
I don't want to over-simplify things and I am replying from cellphone, so
plz keep it in mind.
To make it simple - motor operates better with more power.
Power is voltage times current.
Current is a division between voltage and the coil resistance of the motor.
So, if you want to increase power, you can rewind the motor coil (VERY
difficult) or simply increase voltage.
My motor had a coil resistance of cca 50 ohms.
With 12V I could never get currents above 240mA. Now doubling the voltage
and voila - current doubles too, so the power will be 4 times higher.

pbbob, 2018-01-04:
@sanchosk I am confused on how to connect the tb6612 to the attiny. there are 3 inputs(in1,in2,pwm) for each motor, but only 2 pwm outputs on the attiny. how did you hook it up, and what happened to the leftover pin?

pbbob, 2018-01-04:
@misan the same thing above ^^^^^^ aplies to the a4950. how do you configure all 3 pins? (in1, in2, vref)?

Kadah, 2018-01-05:
@misan The cogging(?) angles of the particular motor and pulley weren't enough to small moves to be a thing with it. I think the best it could do was 45degs, which worked out to like 0.8mm. There was additional backlash in the rest of the system, not just from the DC motor itself. The weight of the print head was also not helping anything.

I found the old video of it printing: https://www.youtube.com/watch?v=RyIZFUmpDHg

I've used both the L298N and TB6612FNG. I mainly used the L298N, but the TB6612FNG had the size advantage and I'd likely start with it next time I try this.

sanchosk, 2018-01-05:
I think I've put the PWM on high (or low, check the specs) and in1 and in2
to ATtiny.

misan, 2018-01-05:
> v=RyIZFUmpDHg

This looks like an HP printer motor. While I do not know their specs, I
reckon they are powered at 38V. If at that voltage then can handle well the
print carriage loads, they may well be underpowered when running at 12V. If
that is the case, it does not matter what h-bridge you use, they will still
be underpowered.

That said, any real-world motor will experience a certain dead-zone (where
a low input value will cause no motion whatsoever).

Cool video though.

sanchosk, 2018-01-05:
My exact experience with the HP motor and carriage - at 12V it was quiet,
but the precision was bad. At 24V it got much better, but the motor drive I
preferred (TB6612) only worked with 12V.

pbbob, 2018-01-05:
@sanchosk the datasheet says that pwm1 and pwm2 have "200kΩ pull-down at internal ". does that mean i should pull them low?

sanchosk, 2018-01-05:
Hmmm, to be honest, I don't remember :( Just test it ;)

pbbob, 2018-01-05:
I am designing an arduino mega shield so that is compatible with marlin firmware for a reprap.

Kadah, 2018-01-05:
@misan Yup, its an HP one but it was a weird 13.4V printer I believe. I kept the plug pack from it somewhere, figured it make a good source for a 12V linear reg if I ever needed it.