misan/dcservo #41: attiny85 pins

pbbob, 2018-01-02:
does motorpwm1 and 2 go to the h bridge? also, how do i get the P, I, and D values for tuning? why are there 2 encoders(a and b)?

misan, 2018-01-02:
OPTION 1: Motor pwm1 and pwm2 go to In1 and in2 signals of an h-bridge (which is supposed to be enabled).

OPTION 2: Alternately, you could use in1/in2 pins for direction and a single PWM signal on the enable pin.

---------
1) P,I and D values can be set in the firmware initialization.
2) PID values can be queried and modified using the serial port and even stored in the EEPROM.
3) If valid PID values are found stored in the EEPROM, then these would be used instead of the firmware ones.

---------
Encoders have two signals so not only steps can be measured (angle) but also the direction of it (sign).

pbbob, 2018-01-04:
how do i determine the pinouts for a 6 pin encoder?

pbbob, 2018-01-04:
should there be a resistor somewhere beween the encoder pins and the attiny85?

sanchosk, 2018-01-04:
Check this video:
https://m.youtube.com/watch?v=a55EwMSIy2w

sanchosk, 2018-01-04:
I strongly recommend to tune and setup your encoder+motor driver using
arduino first and only when fully working and pid id ntified, upload the
final version to attiny. Due to lack of pins on the package, re-programming
is quite painfull and you will have to use 12V reset every time.
These days you can get full arduino nano board for pennies from China that
saves a lot of headache comparing to attiny...

pbbob, 2018-01-04:
what is the best h bridge to use, and also, where can i find a good tutorial on how to use h bridges/how they work?

pbbob, 2018-01-04:
@misan "Encoders have two signals so not only steps can be measured (angle) but also the direction of it (sign)"
when connecting encoder a and encoder b to the attiny(or arduino)am i correct that it doesnt matter which encoder is connected to atttiny 3 - encoder A and attiny 4 - encoder B?

misan, 2018-01-04:
Have a look at these blog entries
http://www.modularcircuits.com/blog/articles/h-bridge-secrets/h-bridges-the-basics/

misan, 2018-01-04:
No matter how careful you are, there are two cables that can be inverted:

DC motor power wires

Encoder AB wires

My advice is to connect them first at random and check if servo seems to
want to keep the motor shaft at the original location (it fights you back
if you try to move the shaft by hand). If that is the case, then please
write down the right color/signal for each pin.

But 50% of the time following the above advice will lead you to your motor
rotating full-throttle and never stopping. This is because there encoder
and motor are not in phase. Simpler solution? just swap motor cables. And
now you should see the servo works (if not there are some problems with
your hardware).

pbbob, 2018-01-04:
@misan so it is okay, or not okay for the dc motor power wires and the encoder ab wires to be inverted?

misan, 2018-01-04:
You need to figure that out. It depends on the motor.

alifilhan0, 2021-11-13:
Hello, sorry to disturb. I just want to see the possiblities with the ATtiny85, that is how low I can go. Is it possible to use an IR photo transistor(two pins) with the attiny? The phototransistor's off and on state determines the direction/step of the motor. Is the already given code compatible or not?