misan/dcservo #45: atmega8 support..

iramuhammad, 2018-04-26:
I know it's possible to upload the sketch to atmega8, my question is, do I need to modify the code by any means?! 

Thank you for your great effort. 

Here in my town hundreds of printers - (inkjets the most) - are just wasted to the landfills,  I am planning to collect it all, hack it, hook a cheap atmega8 on the two optical encoders along with the DC motors in it and a cheap l293d h-bridge IC and there it is!!  a cheap, fast, closed loop motion system. 

I plan to build cnc plotters out of it for the sake of speed!! 😇😉,  also it will be a charity work,  I will send to each school in town a nice diy plotter to teach the students what cnc technology can offer for us,  it will be a great purpose out of those printers, instead of going to landfills, I hope some one will be inspired by this work. 

Parts:
1 - inkjet printer.  1.qty
2 - L293D h-bridge    1.qty
3 - atmega8 1-2.qty -till now don't know for -sure. 
4 - Arduino UNO (breadboard version along with SD card module)  1.qty
5 - some wood and MDF, nuts and bolts, bearings. 

Price: 
1- an inkjet printer is as I said before in hundreds here in my town,  I can get some for free and some other I can buy at 1-2$ maximum. 

2- i 've studied lot's of printers DC motors,  all of them require less than 600mAh,  so l293d IC is great cheap choice to drive them and it's price here in my town 1.5$. 

3- the atmega8A-PU is 1.5$. 

4- Arduino: a costume veroboard with atmega328P-PU,  and SD card module for portability cost both 5.5$. 

5- the wood blocks or/and MDF is used in small sizes,  I bit I can get those for free or near free and with nuts and bolts sold per 3$ for 1KG, I guess it will be sufficient. 

Total cost will be around/between 14.5$ - 20$ maximum per full working plotter. 

For the h-bridge, I've managed to salvage many from laser printers -95% of laser printers have h-bridge ICs in it- , also available here by ten's, along with a minimum 2 solinoids per one printer, the solinoid well work as pen UP/DOWN mechanism. All this will further minimize the total cost per plotter.  and Time?! I have a lot of it 😇☺. 

Thank you again. 
If my English is bad,  forgive me. 

Ira, from Syria.

iramuhammad, 2018-04-26:
I forgot to mension the power supply,  every inkjet printer have a nice boxed high quality power supply ☺.

misan, 2018-04-26:
Atmega8 has only 8KB of flash and 1KB of RAM. I think you'll need to reduce
the program size to fit in. Same holds for memory size.

If you get rid of all serial commands and configure everything in the
firmware you can slim it down quite a bit but I do not know if it will fit
in 8K. And you may need to program the chip directly without a bootloader
to save another 2KB.

On the other hand, access to pins/interrupts code will need to be revised.

iramuhammad, 2018-04-26:
OK,  I've managed to get STM32F103x8 series uC,  at cheap 3,67$,  I will try to adopt your code to work on these beasts,  I think it will be able to provide control of 3 DC motors.

iramuhammad, 2018-04-26:
I will be using this code :

https://github.com/agnunez/stm32duino_2DCMotor_Encoders

misan, 2018-04-26:
I think so, this one runs at 72Mhz :-D

misan, 2018-04-26:
> It can fit a gcode interpreter also.
> I am aiming to get 0.5m/s. It will work 😉

You can peruse the interpreter from here
https://github.com/misan/ESP-stepper-servo

iramuhammad, 2018-04-26:
This man inspired me in the first place.

http://www.homofaciens.de/technics-machines-cnc-v3-1_en.htm

closed 2018-08-12 by iramuhammad