misan/dcservo #47: Use with Cytron 13A Driver

movik1985, 2018-05-25:
Hello,
First things first, thank you for the brilliant work and making it available for all.
Now, I am thinking of using a larger, more powerful DC motor that may draw a lot more current than L298N can handle. I want to try the Cytron 13A driver. Unlike the L298 it doesn't have a IN1 and IN2 but has a PWM and DIR input. I am wondering if the following modification to the original work.

void pwmOut(int out) {
   if(out<0) { digitalWrite(M1,0); analogWrite(M2,abs(out)); }
   else { digitalWrite(M1,1); analogWrite(M2,abs(out)); }
  }

where M1 connects to DIR and M2 to PWM. I believe it should work but I can't get my head around some other parts of the code to be really sure.

Thanks for your advice.

Best Regards
Vikram

misan, 2018-05-25:
That sounds about right Vikram. Let me know how it goes.

movik1985, 2018-05-26:
Thanks Miguel. I'll give it a shot and keep you posted.