misan/dcservo #52: Can i use a encoder with step only instead of a quadrature encoder

Animalrsa1, 2018-08-06:
Good day, many thanks for making your work public, i was wondering if it would be possible to use a step only encoder (no direction) with your project to control the position of large dc motor, ramps will still send its normal step and dir but the closed loop will only be on steps, is this possible?

many thanks
Tony

misan, 2018-08-06:
Hi @Animalrsa1,

It is not possible. 

A tachometer gives you information about motor speed but not direction.

A position control needs to know the direction of each motion to be able to accurately track motor shaft position. Without that, it won't work.

Animalrsa1, 2018-08-06:
Hi

many thanks for the swift response, if possible could you assist with a cost effective solution to a quadrature encoder. i want to use wiper motors to move my mill.

thanks again

tony

misan, 2018-08-06:
You can get a relatively inexpensive one online: https://es.aliexpress.com/store/product/Encoder-600P-R-Incremental-Rotary-Encoder-AB-phase-encoder-5V-24V-Two-Phases-Shaft/3105031_32851443854.html

PaulRobert1, 2018-10-18:
Hello, lemme say, i tried that, bo matter what you do the motors are slow and will overheat fast, if ur pid is too low they wont move properly, if u go higher so ur motor spins up right, it will overheat . U coud drill holes in the motor and cool it by forcing air trough, but u coud try either try servo motor from say a opel corsa steering or bmw x5 compressor motor, etc. I just got some oldschool dc motors wich have 4 huge brushes and rotor, they have about 500 watt, got them at the junkyard. And i used ibt_2 h bridges. Hope this helps.

PaulRobert1, 2018-10-18:
https://www.ebay.co.uk/itm/AEDR-8300-Series-Encoders-Reflective-Surface-Mount-Optical-Encoder-2-channe-W1V8/123276004045?epid=8006938605&hash=item1cb3d28acd:g:XcwAAOSwNDVbW0sH

PaulRobert1, 2018-10-18:
Cheap encoder link, u have to make the disc, i made it with a 3d printer

PaulRobert1, 2018-10-18:
And mirror on the other side, and disc must b thin

Kadah, 2018-10-19:
If I'm reading the datasheet right on the AEDR-8300, the code wheel will need to be 8.35 lines/mm (212 LPI), which seems like it would be somewhat difficult to print.

PaulRobert1, 2018-10-20:
You could go with as many lines as u can, like i did to test and get used to the whole thing until u scavange the real deal :) here s another link to a encoder

PaulRobert1, 2018-10-20:
https://www.ebay.co.uk/itm/NEW-5V-24V-Encoder-600P-R-AB-2-phase-6mm-Shaft-Incremental-Rotary-Encoder-WD/292424483269?epid=11011024104&hash=item4415db81c5:g:jWUAAOSwyH9aKQVL

PaulRobert1, 2018-10-20:
U got to open it, inside u will find 2 empty pads ready 2 solder components, solder 2 4k7 resistors on the pads before giving it a go.

PaulRobert1, 2018-10-20:
With 3d printed encoder disk i managed to get a precision of 0.1, wich isnt that bad taking into account the disk precision

PaulRobert1, 2018-10-20:
Or go to the scrap yard and get encoder strips from old plotters

patracy, 2022-06-30:
I believe I have the same issue.  I stumbled upon this project in the hopes that the controller could be used to build a controller system for a Pegasus robotic arm.  The vendor that made the arm (it's a "educational robot") isn't much help as they won't share any specifics about the computer control setup (which I'm lacking here).  I picked up a nvcmv2.1 and was able to drive the motor on the x axis forward and backwards.  The problem is the encoders on these motors are a very fine incremental/step encoder rather than a absolute.  The motors themselves have easy access to the encoders.  It's just a set screw that holds the disc on.  And two screws that hold down the encoder.  Is there a encoder I could retrofit onto the backend of the motor to convert it over to a quadrature encoder?  I saw a link above for one, but that's really a sealed unit that couples to the shaft.  I was hoping for a simple disc and new sensor to convert it.  I can get more measurements of the disc diameter and the mounting holes of the sensor.  The shaft is only about a 1/8" diameter or a little more.

misan, 2022-07-01:
@patracy: an incremental encoder usually provides a quadrature output, that can be used to obtain an absolute position if you pair that with a way to detect a reference [home] position when booting up. The encoder you linked is a low-resolution unit (only 12 pulses per revolution), it is an incremental encoder, and it provides a quadrature output too.

Alternatively you can use Magnetic encoders to obtain an absolute position (angle), like the AMS5600: https://ams.com/as5600

patracy, 2022-07-01:
I have tried the dcservouno sketch.  But it acts very strangely.  The motors I have have a HEDS-9000 encoder on them.  They have a 500cpr on them.  When I power up the arduino I get nothing out of the serial port.  But when I disconnect the encoder A/B leads, it will then provide the menu.  I can then reattach the AB leads and I am able to home it.  Set to T mode and it will randomly move as expected.  I was having jitter at home when I first started it but I put P down to 1 and it seems to be behaving now.  Is this encoder setup going to be too much for the mini as far as counting goes?  I know it's going to be excessive since it's a gearmotor setup on the number of counts.

patracy, 2022-07-05:
I think maybe the encoder setup is just simply too much for this.  It's 500cpr on the motor shaft side of things, feeding a gearbox.  I bought this cheapo DC motor/encoder just to try it out and it behaves just fine it would seem.  I'm going to try a hall effect encoder like it has on the style motor I have and see if that would help.  

https://www.amazon.com/dp/B0833X3KY3?psc=1&ref=ppx_yo2ov_dt_b_product_details

misan, 2022-07-05:
You can always not power the motor and move the motor shaft by hand (while
the encoder is powered and connected). The "?" command on the serial port
will show you whether the location changes or not (useful for
troubleshooting).

Please note the motor you linked has a low-resolution encoder, which is not
a problem. but due to the gearbox you should get a nice resolution. However
... the encoder measures the motor shaft and NOT the output shaft, which
means that any amount of play in the gearbox can ruin the accuracy of your
measurement anytime there is a direction inversion.

misan, 2022-07-08:
The checksum error is about the EEPROM storage not being initialized. Once you use "W" command this should go away.

patracy, 2022-07-08:
It's unresponsive when I try to type anything.  The L led goes solid.

misan, 2022-07-08:
when using an I2C encoder if it becomes unresponsive it seems there is a
problem with I2C communication, you should check your circuit with i2c scan
code you can find online.
https://create.arduino.cc/projecthub/abdularbi17/how-to-scan-i2c-address-in-arduino-eaadda

misan, 2022-10-11:
Please note the Arduino cannot handle very high interrupt rates, so it may
only work for low speeds if you have a high-resolution encoder. You may
want to lower the motor voltage to see if you can move within a low-speed
range for it to work.