misan/dcservo #53: PId Tuning tool not displayng graph correctly

PaulRobert1, 2018-09-25:
Hello, 

I am currently working on modifying a old school milling machine, modifying it using grbl and your amazing code for the servos, I am using windshield wiper motors to move the axes, encoders from china and 3d printed encoder discs but i have hit a sinkhole:) these are the issues I am currently facing:
- cant get the pid tuning tool to display correctly the graph( it shows only on the 1st square of the whole graph and slowly moves down, motors overheating from bad pid, and after moving say x axis, after it stops, starts jogging back and forward.
- when lifting z all axes are going crazy
I am using 3 arduino nano for driving 3 IBT_2 H bridges and a arduino nano running grbl.

Thank you very much for all the support provided.

Kind Regards,
Robert.
[image]
[image]
[image]
[image]
[image]
[image]

PaulRobert1, 2018-09-25:
OMG! that was a unexpected fast response:)

I am powering my milling machine from my Off grid Photovoltaic system, from the batteries more exact, that s 600amp hours on 24v.
I am gonna give it a try with P0.1,I0,D0 see what happens.(everything slowed down:-) )
I can see what u mean with the encoder after the gearbox, backlash is what I had in mind:) and I made sure that the gearbox from each motor diddnt have any noticable play on it

I uploaded 2 videos to google drive with videos of the pid tuning tool and z axis erratic behaviour
https://drive.google.com/open?id=1jRwQhsOKyt0WeZdFfnS9Pzzm7uZ7SPew
What are your toughts on z going down smooth and when lifting z all axes going crazy?
Thank you.

misan, 2018-09-25:
Well, if powered from batteries I guess juice is plenty (only if a too thin cable is used lack of power might be an issue).

PaulRobert1, 2018-09-25:
calbe is 4mm(diameter) silver coated they are rated at 70 amps, but the machine goes crazy only when i lift z:))) when i move x or y everything works well, check the video out when i lower z everything is smooth, when i lift z movement is not linear at all

misan, 2018-09-25:
I saw the videos, Z axis up is surely creating some noise that seems to affect the rest, I do not see why noise and activation seems quite different than when going down. It may well be the motor's dead zone. Something I have not yet addressed in my project.

PaulRobert1, 2018-09-25:
coud it be encoder related( i have 2 adjustable resistors i think that s what they are called in english)

PaulRobert1, 2018-09-25:
gonna try that too

PaulRobert1, 2018-09-25:
Thank you very much for taking the time to help me out, i really diddnt expect such a fast and helpful response:D

misan, 2018-09-25:
It is the least I could do considering the love you have put to the project

PaulRobert1, 2018-09-25:
Thank you 4 the kind words Misan:)

I have removed the z axis letting the motor run freely, and the erratic behaviour stops but starts as soon as i apply a little bit of force on the motor (bad brushes maybe) taking into account that these motors are from a salvage yard:) gonna put a new motor in, see what happens would ceramic capacitors on the motor leads help in filtering the ripple that is going on there?

PaulRobert1, 2018-09-25:
also tried switching polarity to the motor, same result, diffrent direction lmao

misan, 2018-09-25:
maybe you can move z motor to x or y where hopefull less torque is needed
... ?

PaulRobert1, 2018-09-25:
actually the z motor wich has screw and nut for movement requires the least torque, the other axes require 5 nm just to start moving

PaulRobert1, 2018-10-11:
Hello misan, i managed to get everything working, the noise was coming from a motor winding wich was tied to ground, fixed it, now I am still stuch at the PID tuning tool not displayng graph correctly:(( How can I fix that?

misan, 2018-10-11:
Not sure what the problem could be. The graphical tool does issue an "S"
command afer the motion. There are some built-in delays in the code that
usually work, but if your system is too fast or too slow they might need to
be adapted. The goal is for the data transmitted by the S command to fully
finish before the graph is drawn.

Unfortunately, there is not much RAM on an Arduino, so only 999 points can
be recorded. If memory serves, I think at 1ms per loop that is just 1
second for the whole move. If the sample move takes longer than that there
is no way the current code will draw the full graph. You could try reducing
the counts of the move from the existing value to its half to see if that
helps.

PaulRobert1, 2018-10-11:
Thank you very much, i am gonna try that, i l let u know how it works.

PaulRobert1, 2018-10-11:
what step time and step pulse time do u recommend in ns?

misan, 2018-10-16:
@PaulRobert1 I expect step pulse time longer than 1uS. Not sure about the minimum acceptable time though.

closed 2018-10-18 by PaulRobert1