misan/dcservo #54: Pro Micro pin reassignment?

paukstelis, 2018-09-25:
Bit of a noob question: I want to free up pins 2 and 3 on the Pro Micro for I2C. If I put encoder interrupts on Pin 1 and Pin 7, what would need to change in encoderInt function? Port manipulations scare me.

paukstelis, 2018-09-26:
Hmm, according to the sparkfun page, ProMicro has 5 external interrupts. Shouldn't it be possible to just swap them? Is this accurate?:

> "The Pro Micro has five external interrupts, which allow you to instantly trigger a function when a pin goes either high or low (or both). If you attach an interrupt to an interrupt-enabled pin, you’ll need to know the specific interrupt that pin triggers: pin 3 maps to interrupt 0, pin 2 is interrupt 1, pin 0 is interrupt 2, pin 1 is  interrupt 3, and pin 7 is interrupt 4."

misan, 2018-09-26:
That is correct but they do not use the term INT4 or INT5

[image]

misan, 2018-09-26:
And yes, they can be changed, just remember to update any PIN* code accordingly. The image below shows the port and number for each pin:
[image]

misan, 2018-09-27:
CHANGE=RISING+FALLING

more details https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

You can select HIGH instead and use it for the STEP input.

closed 2018-11-02 by paukstelis