misan/dcservo #56: Using lpd3806-600bm

PaulRobert1, 2018-10-18:
Hello misan,

Is it possible to use a lpd3806 rotary encoder with your code?

Thank you.

misan, 2018-10-18:
Sure, no problem other than the coupling with the motor.

PaulRobert1, 2018-10-18:
Couse i did and the motor goes crazy like it would not detect it, tried switching a with b, mayb raise voltage? In the datasheet it states 5-30 vdc 
You are a patient man! :)))

PaulRobert1, 2018-10-18:
I powered it from the arduino internal 5v stabiliser

misan, 2018-10-18:
@PaulRobert1 What you need to be sure is the input pull-ups are enabled as the outputs of the encoder are open-collector according to this tear-down https://wemakethings.net/2014/05/26/rotary-encoder-teardown/

Or you just place a 4k7 resistor on each encoder input.

PaulRobert1, 2018-10-18:
Gonna do that, thank u :))

closed 2018-10-19 by PaulRobert1

misan, 2018-12-09:
> Hello, long time:)
>
> Had time and made progress, and stumbled upon a issue, the x axis wont
> move only in 1 direction, won t reverse...other axes are ok, and same code
> and wiring. Any tought that comes to your mind what the issue coud be?
>
> Kind Regards.

If you are using it with Marlin that sounds like an endstop problem.

PaulRobert1, 2018-12-09:
I am using it with LinuxCNC.

misan, 2018-12-09:
Not sure about it: Controller should honor Direction pin. If it does not,
on a single controller, I would double check the signal is, in fact,
changing value and, if that is the case, verify wiring.

If all looks ok, I would check the controller in stand-alone mode (use
serial comamnds to check motor moves in the two directions). If all works
ok then I would check if direction pin in Arduino might have been damaged
(does it still can read a proper digital value on than pi, both low and
high values?).

PaulRobert1, 2018-12-09:
Problem was faulty wiring on the x dir input pin, all working as should.

Thank you!