misan/dcservo #57: Backlash correction?

paukstelis, 2018-11-02:
I'm using geared motors and there is a reasonable amount of backlash. What is your thought on where best to implement this? It seems that we could fairly easily track direction from the pwm output and just look for cases when direction changes and modify the target position with +/- the number of encoder steps needed. This seems too easy, so I'm sure I'm missing some critical factor here!

misan, 2018-11-02:
That sounds about right. It may be useful to have the compensation value
stored in the EEPROM so you can change it without recompiling the code.
However I am not sure how well that will work (but I would like to know if
you follow that route).

paukstelis, 2018-11-08:
So it seemed simple, but becomes more complex in practice. I was basically doing what I described: tracking direction and applying modification to target when direction changed. The problem became that any overshoot (which I can't totally avoid because of large masses) would trigger a correction, which leads to stuttering. I'll have to think if there might be a better way to do this.

paukstelis, 2018-11-08:
Oh, I think the fix is simple. I was monitoring motor direction via the sign of the PID pwm output. I can just watch when DIR changes. That should do the trick.

misan, 2018-11-08:
Proper adjustment of the PID gains could give you a response without overshot

paukstelis, 2018-11-08:
I've never been able to get it spot on. The larger the motors (and the higher the operating voltage) make it more difficult to tune, from what I have seen.

I tested watching changes from the DIR signal and that seems to work well if you assume isotropic backlash. The only complicating factor is that on one axis I have a somewhat restricted drag chain that soaks up some of that backlash depending on where it is. I'm not sure I want to get in modeling this to compensate.