misan/dcservo #60: PINF  question (error)

caslor, 2019-04-26:
Hi
i want to experiment with a arduino pro mini (or nano) with this code
so when i am using the dcServoProMicro.ino  to compile  i am getting this error :

`void countStep(){ if (PINF&B10000000) target1--;else target1++; } // pin A0 represents direction == PF7 en Pro Micro
                       ^
exit status 1
'PINF' was not declared in this scope` 

Ok this is because on arduino pro mini the pin ports registers are different 

for  AO pin (PINF  in ProMicro)  in   ProMini would be  PINC  

so when i change that to the code i can compile without any arror

my question has to do if it is worth it to try that code with an arduino pro mini  for  low speed motor

misan, 2019-04-26:
Hi Pro micro and Pro mini are different boards using different processors.

Pro mini can work but you need to use the code for Arduino UNO and not for
Arduino Pro Micro.

I reckon you already cracked the port assignments
[image: image.png]

caslor, 2019-04-26:
Thanks for the quick reply 
so will make my try with the   dcservo.ino

thanks again for the suggestion... have a nice day :)