misan/dcservo #62: Question: Servo logic vs H-Bridge?

cbpdk, 2019-07-25:
I have a continious rotating MG395 servo connected to a Freescale 100 slot disc and optical quadrature encoder module.
A near destructive hw mod rotating the optical encoder module 90 degrees was needed before it acted as a quadrature encoder.
Using the encoder gives a resolution of 360/400 = 0.9 degrees.

The MG395 only contains a servo control board.

The specs say it moves 60 degrees in 180 ms without load.
However, the servo has a deadband around the 90 degrees signal, approx. +/- 15 degrees.

For tests I am using an Uno and a sketch based on a modified verion of 
dcservo_magencoder_servo360_MKR1000.ino

This to quickly adjust PID parameters and view the position errors using a serial connection.

Is there any advantage replacing the servo logic board with a h-bridge(like TB6612FNG) with regard to position precision?

misan, 2019-07-25:
I do not think so.

cbpdk, 2019-07-25:
Thank you very much, saves me for the trouble of testing to no avail.

closed 2019-07-25 by cbpdk