misan/dcservo #63: PID tunnig tool not read the serial port

SirGioVan, 2019-09-08:
The PID screen stays on gray `window`

[image]

[image]
 .

misan, 2019-09-09:
You need to figure out which one of the listed ports COM3 COM4 .. COM11 is
the right port for your arduino. Latter, on line 26 you need to set the
right number to where now says [4]. It seems from your screen that it is
COM4 so you need to set it to [1] (as COM3 is [0]).

Give it a try.

misan, 2019-09-09:
or you can just make portName="COM4";

SirGioVan, 2019-09-11:
> ys [4].
Apenas asi loqre que se mostrara en la consola que se conecto a la IP 192.168.1.239 , pero el motor no se mueve se queda estatico , pero si me conecto con Putty si acepta los comandos, estoy usando el ESP8266 
[Diagrama.pdf](https://github.com/misan/dcservo/files/3599288/Diagrama.pdf)

[image]

misan, 2019-09-11:
Serial connection is established, not sure what is wrong with it.