misan/dcservo #66: Can I use this for a reversible AC motor?

kjjordans, 2020-04-17:
I have a Chinese laser cutter with an AC motor like this that controls the up and down movement of the bed. 

[image]

It is controlled by two momentary switches for one forward and one reverse. 
I thought that I could use dcservo with two solid state relays to control the motor. 

[image]

Am I out of my mind or do you think that it will work?

misan, 2020-04-17:
Yes, you can, but you will need to add a variable frequency drive (VFD)
which will make it an expensive proposition.

Alternatively, you could use vESC or ODrive projects if you remove the
capacitor from your motor, as it seems a regular brushless motor. None of
these are cheap either, but they can do the job.

kjjordans, 2020-04-17:
Thanks for the quick response.
The motor is quite slow so I don’t think I need pwm. Just on and off would be enough. It might be easier to swap the motor for a stepper

misan, 2020-04-17:
It all depends on the error you can accept. You can try with a dead-band to
prevent constant oscillation (so the motor is not activated until the
output signal is higher than a given threshold).