misan/dcservo #67: ESP8266 NODEmcu V2 connection issue

ravx4, 2020-04-23:
Hello Mr.Miguel,
I'm having a little fight with ESP. At first, when I tried to upload sketch "dcservoESP_wifi.ino" I've encountered an "isr not in iram" error. After googling it, I've downgraded in board manager to esp8266 v2.5.0 and added
`void ICACHE_RAM_ATTR ISRoutine ();`
to your code, the sketch did upload and in serial monitor I'm getting:
> Connecting to Patrol 2.4

> .

> WiFi connected

> Server started

> 192.168.0.101

Is that ok? how can I input PID values? or read encoder position?
If there's any other info needed to solve this issue, I'd try to provide it.
I'd be very gratefull for any help from You :)

misan, 2020-04-23:
Hi,

I reckon the menu was not shown in the TCP stream but it was showed over the serial connection. At any rate, "h" command will show you the available commands, that will show you how to change PID values, move the motor, etc.

ravx4, 2020-04-23:
Hi, tkank you for reply, I'm typing "h" in serial monitor but nothing shows up

misan, 2020-04-23:
> Hi, thank you for reply, I'm typing "h" in serial monitor but nothing
> shows up

Hi,

Assuming you are using
https://github.com/misan/dcservo/blob/master/dcservoESP_wifi.ino code then
I was wrong: help will be shown over the TCP connection.

connect using, for example, with: nc 192.168.101 23
and type h for help

ravx4, 2020-04-23:
I've found solution by to your advice :D I've downloaded SocketTest v3.0 to connect via wifi (I've been trying to connect via USB) And now i can see and adjust PID. Just one more question:
> Please note EEPROM storage code will not work in this platform :-(

storing values of PID to ESP requires to flash it with sketch with manually written kp,kd,ki?

misan, 2020-04-23:
Yes, the ESP8266 does not have a built-in EEPROM as the Arduino does, so I
ditch that functionality (that it could be obtained with some extra coding
by using some area of flash memory for that). However, my experience with
the ESP8266 is that PWM output is not reliable all the time, so from time
to time, there will be a spike on the motor. That for me was a deal-breaker.

ravx4, 2020-04-23:
I'll experiment with ESP anyway and see if I get the same spike on PWM ;) i can always switch to pro micro as I planned at the beggining. 
By the way, could you confirm my understanding in connecting ESP to IBT-2 (BTS7890) module?

>const int M1=16;  goes to R_ENABLE
const int M2=5;   goes to L_ENABLE
const int PWM_MOT=15;   Goes to both R_PWM and L_PWM

or

> const int M1=16;  goes to R_PWM
const int M2=5;   goes to L_PWM
const int PWM_MOT=15;   Goes to both R_ENABLE and L_ENABLE

i dont want to risk burning IBT-2 module :(

misan, 2020-04-23:
I would go with the second wiring:

const int M1=16; goes to R_PWM
const int M2=5; goes to L_PWM
const int PWM_MOT=15; Goes to both R_ENABLE and L_ENABLE

ravx4, 2020-04-23:
ok, I'll give it a try. Anyways, I hope that this setup would work on my D-Bot :)