misan/dcservo #69: Code Modification

TheGermanIng, 2020-05-03:
here you could give suggestions for modifications. I will start with:

// chose one
TCCR1B = TCCR1B & 0b11111000 | 1; // set 31Kh PWM
TCCR1B = TCCR1B & 0b11111000 | 10; // set 8Kh PWM
TCCR1B = TCCR1B & 0b11111000 | 11; // set 1Kh PWM

// Step in Multiplicator example x10 future: better you found a menu eeprom option for this
void countStep() {
  if (PINC & B0000001) {
   target1 = target1 - 10;  // hier die anzahl der encoderschritte / impuls
    directionLast = -1;
  }
  else {
    target1 = target1 + 10;  // hier die anzahl der encoderschritte / impuls
    if (directionLast < 0) {
      target1--;  
      directionLast = 1;
    }
  }
} //

misan, 2020-05-03:
Hi,

I discarded audible frequencies for the PWM signal, as the motor would be noisy. Of course, the higher the PWM the frequency the higher the switching losses on the h-bridge.

I do not understand the Multiplicator thing: what is the goal?

TheGermanIng, 2020-05-04:
I have different power amps, 31khz is really too high for almost all ic, 10-20 khz is good, you probably have to set the PWM to 9 bit to get there. I use bigger and bigger engines up to 2kw has already been there.
The motors also always have a gearbox, now calculate example encoder 360p / T, most sources make up to max 20khz mach3, ramps without stumbling, your software double evaluation is 10khz encoder, i.e. max 1600rpm after a 1:50 gear 32U / min, with a step doubling that would be 64 rpm, now I don't always have the 20khz source can sometimes only be 2khz, but the speed needs anyway, the accuracy does not matter according to the gearbox,

TheGermanIng, 2020-05-04:
a cheap washing machine motor dc, creates more than 500W, 10,000 rpm, an h-bridge with 300V does not create 31khz, the sound doesn't matter if the machine works, but it mustn't have any drift, my big machine weighs 3 tons, what do you think you if a 100 mm SK40 end mill lands a little off the position?

misan, 2020-05-04:
Of course, there is not such a thing as one-size-fits-all. So feel free to use a frequency that works with your system. That does not interfere with the software at all.

If you have a gearbox, then the noise it creates is most likely going to be more important than the sound of the coils due to PWM frequency. 

If you have a spindle and a tool milling, then that most likely will be the main source of noise :-)

But remember that together with a gearbox it is going to come some play. 

Besides, if using a gearbox you might want to use the encoder in the output shaft instead of in the motor shaft. This way you reduce the frequency of the interrupts and measure the true angle of the output shaft regardless of the play in the gearbox.

Of course, the whole purpose of a closed-loop is to get accurate positioning (in this case), so drift is not acceptable. I did not have drift under normal circumstances with the setup I used, but I am always warning users their Arduinos will not handle more than around 60.000 interrupts per second. If that is your case you should consider using a faster processor, like an STM32.