misan/dcservo #70: Step multiplier

just-jason, 2020-06-20:
The max step frequency is often a limiting factor with 8 bit controllers. When using geared motor setups for the servo it is not uncommon to end up with an unnecessarily high encoder step count if the mounting of the encoder on the gearbox shaft is not possible or desired, but rather only on the motor shaft. Sending the required high step signal rate becomes problematic with higher speeds for a slower 8 bit controler. Would it be possible to introduce something like a step multiplier so that each step signal sent by the controller is treated as a signal to move x steps by the servo. I would expect a range of integers between 1-10 should probably cover most cases. The loss in accuracy would in my case be negligible. The mechanical accuracy and stability are never going to be comparable to the theoretical accuracy achieved when fully counting each of the encoder pulses.

misan, 2020-06-20:
Hi @just-jason, 

A step multiplier could be something easy to do, but ... a motor with a gearbox might need to turn at high speeds to get the job done. And while doing so it might create a high rate of encoder pulses that, again, might be a problem for an 8-bit microcontroller. 

So while the multiplier would reduce the number of step pulses needed (thus reducing the microcontroller interrupt rate), it will not avoid the high rate of encoder pulses (which may still be a problem).

Given that nowadays a 32-bit microcontroller is  cheaper, faster, and more energy-efficient than older 8-bit microcontrollers, I would go that way whenever the Arduino-based solution is not fast enough.

just-jason, 2020-06-20:
After writing this request I discovered the blue-pill fork of this project. It now seems even more useful to incorporate this feature as the STM32 could be taking care of the high number of encoder pulses while the main controller will be handling multi axis timing issues and is off loaded from the high step rates. It would follow the logic of target control, ie off load the main controller

RudyFiero, 2020-06-21:
_**After writing this request I discovered the blue-pill fork of this project.**_ 

Could you provide a link to that? I have an encoder that give 2000 pulses per rotation and I would like to try something faster than the pro-mini.

ithinkido, 2020-06-21:
@RudyFiero Sure,
https://github.com/paukstelis/dcservo
From what I can see it should be able to handle three motor+encoder servos with one board !!

just-jason, 2020-06-21:
@RudyFiero
Take a look here https://github.com/paukstelis/dcservo
From what I can see it should be able to handle 3 seperate encoder+motor setups on one board !!
Could I ask what controller you are using to send step pulses to your current pro-mini servo setup?

RudyFiero, 2020-06-21:
@just-jason  I haven't got that far yet. I have some servo motors from a pick and place machine dismantled. I am going to use this project to see what I could do with them.