misan/dcservo #73: Free nano shield blueprint

MarcinanBarbarzynca, 2020-08-14:
Hello. I prepared small 1 sided shield for DCServo nano. 
Please check if its compatible with your scetch. I want to add small OLED SCREEN for debuging purpose. 
https://easyeda.com/polaski/arduino-nano-dcservo-emulator
There is some points added to:
D6 is VCC and D11 are GND now
#define GND          11
#define VCC       6

Please check that i didn't made any mistake 👍

MarcinanBarbarzynca, 2020-08-14:
https://easyeda.com/polaski/arduino-dcservo-emulator For uno

misan, 2020-08-16:
I guess you will use D6 to power the encoder Vcc. If so, remember to set it as an output and to HIGH in your setup.

While you can add an OLED screen for debugging, remember you can too use the serial connection for that same purpose.

MarcinanBarbarzynca, 2020-08-16:
https://easyeda.com/polaski/esp32-cnc-shield_copy
OK. So lets pair it with that shield :P

MarcinanBarbarzynca, 2020-08-17:
How does that Homing pin works?

misan, 2020-08-17:
> How does that Homing pin works?

Travel limits can be detected automatically, for example, during start-up. The motor can turn freely till encoder counts stop, possibly signaling the end of travel in that direction. That could be set to zero, or even a slight negative location. From then on, every time the motor is commanded to zero location (or below) a "virtual" home signal can be generated. That can come in handy whenever you want the controller to operate similar to a stepper motor with an end-stop switch. 

That additional pin is an output from DCservo that will be connected to the end-stop input of a motion controller (running, for example, Marlin firmware) for seamless integration.

MarcinanBarbarzynca, 2020-08-17:
In my case it may signal "I can't keep up" to let the main grbl driver know that he need to w8 with execution. 
I've got 2 PNP limit switches on every axis, so that homing pin will tell me if there is problem with friction. 

I think there may be a problem with that zeroing. When i run test for 1st time it start counting from >15000 steps and by rotating it decrease to 0. AHHHH i see that option in code xD, Homing on reset is optional :) 

Ok. Is homing pin up if motor can't keep up? Am I able to set threshold for that signal?

misan, 2020-08-17:
It is possible to generate two signals, one for MIN and another for MAX travel. I just did not bother with the MAX travel signal.

My initial idea was that this "virtual" outputs will be triggered by the encoder count values and not due to the amount of the position error. But that is open for changes in case you want to go that way. And if you do, then a settable threshold may come in handy.

That said, I did use the position error initially to set the encoder value where the limit should be.

MarcinanBarbarzynca, 2020-08-17:
Yea, i want to go for it. Becouse it may be crucial feature in my CNC build. It let me calculate max feedrate i can go. 

Can You rethink how to add that to code? 

Ok. Can You kick out the homing procedure and add setable threshold when the "Homing_pin" is UP?

In my imagination it looks like this.
Im using 2400PPR encoder. My motor can't rotate by 2 degrees (prbbly 13 steps) and I get "Homing" pin HIGH.

misan, 2020-08-17:
But position error higher than a threshold is a condition that may happen anywhere across the travel of one axis (if enough force is applied opposite to the motion). So that may better be an ALARM signal than a marker of the end of travel ...

MarcinanBarbarzynca, 2020-08-17:
Yes. You got the point 
Alarm signal for main GRBL board to slow down execution.

misan, 2020-08-20:
It could be something like this located at https://github.com/misan/dcservo/blob/master/dcservoUNO.ino#L239

`if (abs(setpoint-input)>LIMIT) digitalWrite(ALARM,HIGH); else digitalWrite(ALARM, LOW); `

Making it part of the endstop code makes sure it is run every cycle the PID is updated (typically every millisecond). 

Of course the value for LIMIT could be set using and additional command over the serial line and the ALARM pin would need to be assigned to a free I/O configured as an OUTPUT.

MarcinanBarbarzynca, 2020-08-20:
Yea. This should be good. Arduino usb is enough to setup this. Users need to setup PID values as well as Alarm threshold.

MarcinanBarbarzynca, 2020-08-20:
Can You add this?

misan, 2020-08-20:
Actually, homing is not exclusive to this feature, you can have both. The point is that I am not totally sold on how useful that could end up being. So I was expecting your feedback once you have it working. To get the code up and running you just need to add that code to the line I mentioned above(assuming you are using the UNO version). The ALARM pin is anyone of your choosing and you can use a constant instead of a variable to keep it simple for the LIMIT (as in `#define LIMIT 70` ).

Once you have tested it and verified its usefulness I will be more than happy to add it.

MarcinanBarbarzynca, 2020-08-20:
Ok. Im using uno version. I will test it. Maybe i wont even use homing pin at all... I've got problem with my GRBL driver with step sticks.

misan, 2020-08-20:
Please note that usually, GRBL will emit a short pulse for each step. That
going through an optocoupler may slow things down, so you may need to force
GRBL to use longer pulses than the ones the stepstick would be happy with.