misan/dcservo #78: Help with open collector encoder.

kotopuz, 2020-11-29:
Dear all. 
I got things with small motor. Now i moved to 500w. Accidently I've dropped encoder. After that I position readings are unpredictable. Looks like encoders are very gentle devices. Now I found another with open collector. I cant make it work. Please advise.  Tried to connect it to 5 volts. Looks like I can measure voltage across power plus and output. Nothing from gnd to A or B. Should I add some resistor?  Type is omron e6b2-Cwz6C

misan, 2020-11-29:
The more info you provide, the higher the chances you can get help.

We do not know:

what processor are you using?
what is the code you are running? (are pull-ups enabled in encoder inputs?)
how are you connecting the encoder to the circuit?
when describing your problem, more info is better ("I can't make it work" is not very specific).

That said, a couple of 4K7 resistors from A and B to +5 (assuming that is the voltage of your microcontroller) won't hurt.

kotopuz, 2020-12-02:
Finally I found how to fix it. I was very very lucky that encoder can run at 5 volts which simplifies a lot solution. I've connected 0 and 5 volts to ESP32 Vin and GND. A and B as per sketch. Trick was to use build in pullup resistors. It means   pinMode(encoder0PinA, INPUT_PULLUP );
pinMode(encoder0PinB, INPUT_PULLUP );

Reason is because open collector only connect output to ground, when it's not connected it's floating and pull up to 5V changes floating to TTL level of 5 volts.
If I use 24 VDC as power I need to make voltage divider.
Now everything works and I have working ESP32 with Blynk interface to jog from mobile phone!
Thanks!

misan, 2020-12-02:
You're welcome. If using external pull-ups with ESP, please remember to connect them to 3V3, not to 5V.