misan/dcservo #79: arduino mega

takfa1992, 2020-12-15:
at the beginning I would like to congratulate you on your work.
i tried the "" cdservoMega_magneticEncoder_brushlessMotor "" file on my arduino mega. It does not work. AND THE SERIAL MONITOR IS NOT RESPONDING.
I tried the file "" dcservoUNO "" the serial monitor responds. I turn the motor in both directions the position does not change but the traget only increases. I think the problem is this part of the code
 "ISR (PCINT0_vect) {// handle pin change interrupt for D8
  Old = New << 2;
  New = (PINB & 1) + ((PIND & 4) >> 1); //
  encoder0Pos + = QEM [Old + New];

}

void encoderInt () {// handle pin change interrupt for D2
  Old = New << 2;
  New = (PINB & 1) + ((PIND & 4) >> 1); //
  encoder0Pos + = QEM [Old + New]; ""
could you help me run your code on an arduino mega. my configuration: a mitsumi m21 2r 14 2241 dc motor with its rotary encoder
and I also have dc motors with  linear encoders found in print
[image]

thank you

misan, 2020-12-15:
> i tried the "" cdservoMega_magneticEncoder_brushlessMotor "" file on my arduino mega. It does not work. AND THE SERIAL MONITOR IS NOT RESPONDING.

That is most likely due to a communication problem with the magnetic encoder. Check the connections and pins involved.

The other issue you mention is most likely a hardware problem too. Make sure encoder signals are connected to the proper pins. Please note the UNO code needs to be adapted to be run in a Mega, as it uses PIN* macros for faster I/O access than digitalRead.

takfa1992, 2020-12-15:
the hardware part is correct. Because I tried another code. and the encoder (optical encoder) works fine. its displays the position well. it is necessary to modify the code to adapt it to arduino mega. i tried but its not working. could you direct me to the part of the code to modify. thank you