misan/dcservo #81: 2 pin IR phototransiator in place of 3 pin Receivers with ATTiny85

alifilhan0, 2021-11-13:
Hello, sorry to disturb. I just want to see the possiblities with the ATTiny85, that is how low I can go. I am trying to make an encoder with 2 receivers and 2 transistors each, the disk having 4 divisions(2 whites and 2 blacks) and the motor will have gear reduction to make it more accurate. Is it possible to use an IR photo transistor(two pins) with the attiny? The phototransistor's off and on state determines the direction/step of the motor. Is the already given code compatible or not? Actually it is no big deal to get 3 pin receivers, but I have a lot of 2 pin receivers lying around and I am thinking to put them to use.

misan, 2021-11-13:
You can do that, but you need to check whether all the code can fit on the ATTiny85. OTOH, if you are not going to use the serial communication you can get rid of a significant amount of code associated with that. However, as calibration is needed I think setting it up might be a bit annoying with ATTiny85. At any rate, please note you'll need to adapt the I/O code to the new pins used in your setup.

alifilhan0, 2021-11-13:
In the present code, the Tiny85 supports 2 receivers right? For step and direction? Or only one? And I don't think I need to change the code significantly because I am just going to connect the emitter pin to the Tiny85's already confihured pins in the code(following the schematics in README.md and in the .ino file) and the gnd pins to gnd(as the phototransistor has only two pins). And before I set the PID values so that it doesn't get annoying to program the chip again and again, I will test it with my arduino uno. Is my approach correct?

misan, 2021-11-13:
Yes, it uses 2 control inputs (step and direction) plus two inputs for the
AB encoder. If you calibrate the PID using an Arduino and then use the
values for the ATTiny85 it will be much more convenient.