misan/dcservo #84: Direction in ATTiny85

alifilhan0, 2023-02-19:
Present code in attiny85.ino looks like the direction isn't supported. Is it a problem here? Also the pro micro code looks small enough 6.8 KiB. Can it be/I try porting it? Also why exactly is the direction isn't supported in tiny85? It does say "let's count steps, we'll care about direction later"

misan, 2023-02-19:
Direction pin is pinr 5 in the tiny85 code:
https://github.com/misan/dcservo/blob/master/attiny85.ino#L15

alifilhan0, 2023-02-19:
Thanks a lot for quick response. But what is this?
https://github.com/misan/dcservo/blob/f771ad9d3e37a16a04d61cf8a9b92b1b7620c768/attiny85.ino#L163

misan, 2023-02-19:
Step is detected first, then, depending on the direction input, the target value is increased by -1 or +1
https://github.com/misan/dcservo/blob/f771ad9d3e37a16a04d61cf8a9b92b1b7620c768/attiny85.ino#L177

alifilhan0, 2023-02-19:
Thanks a lot!

closed 2023-02-19 by alifilhan0