misan/dcservo #86: ramps control setting in code?

Mad-Doc, 2024-12-26:
hello

thank's for this code , i am inspired for control servo motor and most improvement and add (current sensing for homing , more stability pid , ect 

but , i dont understand the "ramps control 1.4"

in code 
bool newStep = false;
bool oldStep = false;
bool dir = false;

what is it ?  i not used in code ? 

thank's :)

misan, 2024-12-26:
Hi,

Could you be a bit more specific? What file and line? Thanks.

Mad-Doc, 2024-12-26:
hello 

in all file code , exemple in "DCservouno.ino"

i have this part of code 

//for motor control ramps 1.4
bool newStep = false;
bool oldStep = false;
bool dir = false;

but , is not used in code ? what is it ? 

thank's

misan, 2024-12-27:
Hi again,

I just have removed these three lines. Not the comment nor the two first
booleans make any sense.

Thank you for helping me out to clean up the code.

Best regards,

misan

Mad-Doc, 2024-12-27:
hello

ok, thanks for confirming my doubts about these boolean lines.

thanks for all the work you did on this code!

I am busy reworking it to work on an IBT2, and have a different homing, when the code is done I will put it on my github :)

it uses pidV2, the encoder input receives an ls7141 in order to have a 4X resolution, a part has been modified to have more speed of execution, a setting of the maximum current in the motor, etc., and I remade the "software tuning" program in C to make it a *.EXE program, more complete more accomplished, also to manage the other drivers running on STM32 based on your code ported to stm32, but also rework (this one will be later, still work to do on it)

many thanks

Mad aka Olivier

(ps sorry for my bad english , i am belgian )

misan, 2024-12-30:
I will keep an eye to your Github. It seems you did quite a lot of work.

closed 2024-12-30 by misan

Mad-Doc, 2024-12-30:
yes, a lot of work, but still finished, I will put everything on github soon :)

thanks again